cleanflight/cleanflight #2405: Cleanup function pointer definitions

hydra, 2016-09-15:
@ledvinap said:
> It is much easier to read the code when you typedef function (not pointer) and use * explicitly when declaring variable with function pointer ..

```typedef void (protocolInitFn)(const rxConfig_t *rxConfig, rxRuntimeConfig_t *rxRuntimeConfig);
protocolInitFn *protocolInit;
```

> @ledvinap And possibly even deference pointer in call (*protocolInit)(...); to make it explicit that function pointer is used

@martinbudden said:
> That would be more my style as well. I was trying to follow the style used by the PID function pointer.

I'm in agreement with the above.

There's also various suffixes that are used, the suffixes are likely just noise words and should probaby be removed at the same time.  e.g. `callback`, `fn`, `func`, `handler`.  If any needs to be kept my vote is for `handler` since it is most descriptive and not an abbreviation.

martinbudden, 2016-09-18:
Actually, I prefer the `Fn` suffix - since the declaration is for a function type. It's no more superfluous than the `_t` that is used in `typedefs`. And `Fn` is better than `callback` or `handler` since the item is a function. That function may or may not be a callback or a handler. (As an aside, I think the term callback is overused - quite often people call functions callbacks when they are just functions or handlers - a callback is a specific use of a function).

ledvinap, 2016-09-19:
``` C
find src -type f -name "*.h" -exec grep -nH -E 'Callback|Fn|Handler' {} +
src/main/io/asyncfatfs/asyncfatfs.h:60:typedef void (*afatfsFileCallback_t)(afatfsFilePtr_t file);
src/main/io/asyncfatfs/asyncfatfs.h:61:typedef void (*afatfsCallback_t)();
src/main/io/asyncfatfs/asyncfatfs.h:63:bool afatfs_fopen(const char *filename, const char *mode, afatfsFileCallback_t complete);
src/main/io/asyncfatfs/asyncfatfs.h:64:bool afatfs_ftruncate(afatfsFilePtr_t file, afatfsFileCallback_t callback);
src/main/io/asyncfatfs/asyncfatfs.h:65:bool afatfs_fclose(afatfsFilePtr_t file, afatfsCallback_t callback);
src/main/io/asyncfatfs/asyncfatfs.h:66:bool afatfs_funlink(afatfsFilePtr_t file, afatfsCallback_t callback);
src/main/io/asyncfatfs/asyncfatfs.h:75:bool afatfs_mkdir(const char *filename, afatfsFileCallback_t complete);
src/main/io/serial.h:126:    serialReceiveCallbackPtr callback,
src/main/drivers/dma.h:20:typedef struct dmaCallbackHandler_s dmaCallbackHandler_t;
src/main/drivers/dma.h:23:typedef void dmaCallbackHandlerFunc(dmaChannel_t* descriptor, dmaCallbackHandler_t* callbackHandler);
src/main/drivers/dma.h:25:struct dmaCallbackHandler_s {
src/main/drivers/dma.h:26:    dmaCallbackHandlerFunc*  fn;
src/main/drivers/dma.h:27:    dmaCallbackHandler_t*       next;
src/main/drivers/dma.h:48:    dmaCallbackHandler_t*       handler;
src/main/drivers/dma.h:64:void dmaHandlerInit(dmaCallbackHandler_t* handlerRec, dmaCallbackHandlerFunc* handler);
src/main/drivers/dma.h:65:void dmaSetHandler(dmaChannel_t* dmaChannel, dmaCallbackHandler_t* handler, uint8_t priority);
src/main/drivers/sdcard.h:52:typedef void(*sdcard_operationCompleteCallback_c)(sdcardBlockOperation_e operation, uint32_t blockIndex, uint8_t *buffer, uint32_t callbackData);
src/main/drivers/sdcard.h:54:typedef void(*sdcard_profilerCallback_c)(sdcardBlockOperation_e operation, uint32_t blockIndex, uint32_t duration);
src/main/drivers/sdcard.h:58:bool sdcard_readBlock(uint32_t blockIndex, uint8_t *buffer, sdcard_operationCompleteCallback_c callback, uint32_t callbackData);
src/main/drivers/sdcard.h:61:sdcardOperationStatus_e sdcard_writeBlock(uint32_t blockIndex, uint8_t *buffer, sdcard_operationCompleteCallback_c callback, uint32_t callbackData);
src/main/drivers/sdcard.h:73:void sdcard_setProfilerCallback(sdcard_profilerCallback_c callback);
src/main/drivers/exti.h:38:typedef struct extiCallbackRec_s extiCallbackRec_t;
src/main/drivers/exti.h:39:typedef void extiHandlerCallback(extiCallbackRec_t *self);
src/main/drivers/exti.h:41:struct extiCallbackRec_s {
src/main/drivers/exti.h:42:    extiHandlerCallback *fn;
src/main/drivers/exti.h:47:void EXTIHandlerInit(extiCallbackRec_t *cb, extiHandlerCallback *fn);
src/main/drivers/exti.h:48:void EXTIConfig(IO_t io, extiCallbackRec_t *cb, int irqPriority, EXTITrigger_TypeDef trigger);
src/main/drivers/serial.h:37:typedef void (*serialReceiveCallbackPtr)(uint16_t data);   // used by serial drivers to return frames to app
src/main/drivers/serial.h:58:    // FIXME rename member to rxCallback
src/main/drivers/serial.h:59:    serialReceiveCallbackPtr callback;
src/main/drivers/serial_uart.h:52:    dmaCallbackHandler_t    dmaTxHandler;
src/main/drivers/serial_uart.h:53:    dmaCallbackHandler_t    dmaRxHandler;
src/main/drivers/serial_uart.h:58:serialPort_t *uartOpen(USART_TypeDef *USARTx, serialReceiveCallbackPtr callback, uint32_t baudRate, portMode_t mode, portOptions_t options);
src/main/drivers/accgyro_lsm303dlhc.h:63:// Uncomment the following line to use the default LSM303DLHC_TIMEOUT_UserCallback()
src/main/drivers/accgyro_lsm303dlhc.h:65:   LSM303DLHC_TIMEOUT_UserCallback() function is called whenever a timeout condition
src/main/drivers/timer.h:46:struct timerCCHandlerRec_s;
src/main/drivers/timer.h:47:struct timerOvrHandlerRec_s;
src/main/drivers/timer.h:48:typedef void timerCCHandlerCallback(struct timerCCHandlerRec_s* self, uint16_t capture);
src/main/drivers/timer.h:49:typedef void timerOvrHandlerCallback(struct timerOvrHandlerRec_s* self, uint16_t capture);
src/main/drivers/timer.h:51:typedef struct timerCCHandlerRec_s {
src/main/drivers/timer.h:52:    timerCCHandlerCallback* fn;
src/main/drivers/timer.h:53:} timerCCHandlerRec_t;
src/main/drivers/timer.h:55:typedef struct timerOvrHandlerRec_s {
src/main/drivers/timer.h:56:    timerOvrHandlerCallback* fn;
src/main/drivers/timer.h:57:    struct timerOvrHandlerRec_s* next;
src/main/drivers/timer.h:58:} timerOvrHandlerRec_t;
src/main/drivers/timer.h:105:void timerChCCHandlerInit(timerCCHandlerRec_t *self, timerCCHandlerCallback *fn);
src/main/drivers/timer.h:106:void timerChOvrHandlerInit(timerOvrHandlerRec_t *self, timerOvrHandlerCallback *fn);
src/main/drivers/timer.h:107:void timerChConfigCallbacks(const timerHardware_t *channel, timerCCHandlerRec_t *edgeCallback, timerOvrHandlerRec_t *overflowCallback);
src/main/drivers/timer.h:108:void timerChConfigCallbacksDual(const timerHardware_t *channel, timerCCHandlerRec_t *edgeCallbackLo, timerCCHandlerRec_t *edgeCallbackHi, timerOvrHandlerRec_t *overflowCallback);
src/main/drivers/serial_softserial.h:27:serialPort_t *openSoftSerial(softSerialPortIndex_e portIndex, serialReceiveCallbackPtr callback, uint32_t baud, portOptions_t options);
src/main/config/parameter_group.h:138:    extern void pgResetFn_ ## _name(_type *);                           \
src/main/config/parameter_group.h:139:    PG_REGISTER_I(_type, _name, _pgn, _version, .reset = {.fn = (pgResetFunc*)&pgResetFn_ ## _name }) \
[... cut ...]
src/main/osd/osd_element_render.h:35:void osdElementRender_callsign(const element_t *element, elementDataProviderFn dataFn);
src/main/osd/osd_element_render.h:36:void osdElementRender_motors(const element_t *element, elementDataProviderFn dataFn);
/* vcp removed */
src/main/msp/msp.h:34:int mspServerCommandHandler(mspPacket_t *cmd, mspPacket_t *reply);
src/main/msp/msp.h:42:int mspClientReplyHandler(mspPacket_t *reply);
src/main/msp/msp_serial.h:84:    mspCommandSenderFuncPtr commandSenderFn;   // NULL when unused.
src/main/msp/msp_serial.h:97:extern mspPostProcessFuncPtr mspPostProcessFn;

martinbudden, 2016-09-19:
Aaargh!

ledvinap, 2016-09-19:
IMO `callback` is overused.
`<*>HandlerFn` makes sense when `<*>HandlerRec` exists

hydra, 2016-10-20:
agreed regarding 'callback'.  'Fn' works for me.  PR?

closed 2017-02-25 by hydra