cleanflight/cleanflight #2411: shift of negative value

MauroMombelli, 2016-09-18:
pid_mwrewrite.c:99:45: warning: left shift of negative value

hydra, 2016-09-25:
what is showing you that error message?  can you provide more details please?

MauroMombelli, 2016-09-25:
Its an error mesaage from the compiler, the function may return -1 in case
of error but is not checked before doing the shift

tracernz, 2016-09-28:
It's due to left-shift of signed value being able to clobber sign bit, e.g. https://ideone.com/F2LDH2

martinbudden, 2016-09-28:
Changing:
```
 ITerm = constrain(ITerm, (int32_t)(-PID_MAX_I << 13), (int32_t)(PID_MAX_I << 13));
```
to
```
 ITerm = constrain(ITerm, -(int32_t)(PID_MAX_I << 13), (int32_t)(PID_MAX_I << 13));
```
should fix it. (ie, move the minus sign.)

ledvinap, 2016-09-28:
@tracernz : what you describe is simple overflow. Best explanation/reason for warning I found is here: https://gcc.gnu.org/ml/gcc-patches/2015-04/msg01529.html.

CF is quite HW specific, porting it to architecture/compiler where left shift produces unexpected value is next to impossible. But it would be interesting to mount UNIVAC on quadcopter ;-)

My opinion is that this warning is far too pedantic, just disable it ...

MauroMombelli, 2016-09-28:
This is not true, ARM by itself does not specify endianess, to the point
you can select endianess at compile time; in any case if just moving a sign
let the warning go away I think is better to keep the warning enabled and
move the sign, because next time may be signal a real bug.

ledvinap, 2016-09-28:
@MauroMombelli : I don't see how this is related to endianness.
Arithmetic shift on twos complement machine is well defined, C standard only accounts for obscure architectures where implementation may be nontrivial. No chance to hide a bug here.

There is also corner case with INT_MIN - `-(1<<31)` triggers signed overflow (`1<<31` is negative value), `(-1 << 31)` is OK. But that is unrelated this issue where bounds should be symmetrical

lesto, 2016-09-29:
Sorry you was talking about a system I don't know, I was thinking it was related to endiness.

About the error i don't know the value of that constant, but if you make it negative and then shift, you end up eliminating the sign bit, so the number come back as positive. Like you never set it as negative.(I'm assuming the shift does not move 1 in sign bit, otherwise the problem would be the upper part of the constrain)

MauroMombelli, 2016-09-29:
Sorry you was talking about a system I don't know, I was thinking it was related to endiness.

About the error i don't know the value of that constant, but if you make it negative and then shift, you end up eliminating the sign bit, so the number come back as positive(well I checked, actually is undefined behaviour). Like you never set it as negative.(I'm assuming the shift does not move 1 in sign bit, otherwise the problem would be the upper part of the constrain)

closed 2017-02-25 by hydra