cleanflight/cleanflight #2412: SP-evo and CMSIS

MauroMombelli, 2016-09-18:
hello,

sorry to hijack the issue tracer, but i'm trying to set up a bare project for the SeriuslyProEVO to run some custom experiment on the quality of the sensors, and I can't make it working.

Right now i'm just trying to make the common blinkon GPIOB pin8 (red led, as cleanflight/src/main/target/SPRACINGF3EVO/target.h), so i've taken the GPIO_Toggle example from StdPeriph_Lib_V1.2.3 and compiled it with this makefile (linker file are taken form this project): https://gist.github.com/MauroMombelli/2d34104cb579f85f8296f3cf9affe7c6

the result hex is uploaded with cleanflight configurator.

Any idea what could i've been missing? does it require any special configuration? i see a quartz near the chip but the values are unreadable :/, but probably i'm missing something really basic.

your puzzled,
Mauro

hydra, 2016-09-25:
Hi @MauroMombelli hope you're doing well, alas, as I'm sure you know I'm super busy with cleanflight and don't have time to help on this, sorry.  I'd also prefer it if other contributors could also stay focussed on immediate goals rather than side projects like this.

If you get it working and do anything interesting with it please do share with us via a link from this issue to your code and/or results.

closed 2016-09-25 by hydra

MauroMombelli, 2016-09-25:
i started from scratch and it works, I still want to fix that makefile as so far when i used one command to compile all i got crazy speedup in compilation time.

So far i managed to get USART1 and SPI1 working and communicatng with MPU9250, taking some pieces from ST examples and CF.

I must say you worked a lot on the code, with that scheduler in place I was quite lost and had to relearn a bit, good job!

As i was working on the MPU code, i must say i found it a bit WTF; seems like there is a nice structure common for all sensor tipology (acc_t, gyro_t, mag_t) and this is really nice, what i find strange is that structure get initialized in runtime instead of compile-time with ifdef, or (solution i'm using) provide a common function definition for init and read, and then the .c that implement them is included by the makefile depending on the sensor to use.

But maybe i lost some gotcha and there is a specific reason why is done like this that i don't see, so i would like to have an input on this.