cleanflight/cleanflight #2418: Baudrates below 9600 not supported

flegmatoid, 2016-09-21:
two issues

a)
"Ports" configuration only enables baudrates from 9600bps, whereas GUI enables establishing serial connections starting from 1200bps. This is self-conflicting setup.

b) lower baud-rates are needed for long-range CLI or Telemetry. RX sensitivity is know to peak at minimal baud-rates (1200bps, 2400bps).

hydra, 2016-09-25:
We could certainly fix it but it's not a simple change since it would invalidate everyone's configurations and we would need to do migrations.  No-one else has asked for this previously so I think there's very little demand for it.  What is your use-case?

flegmatoid, 2016-09-28:
[image]
[image]

Hello Dominique,

let the numbers speak for themselves (see the attached). It gives the RX sensitivity tables for the two common telemetry chips (SX1276 and Si4432).

The 1.2kbps (if compared to 250kbps) gives you and extra link budget of 27dBm(!!!).  This is an equivalent of 500 Watt of extra power fed into your Telemetry module. Or, this is an equivalent of going behind the horizon (70+km), if compared to <100 meters. Unfortunately, this configuration is impossible on my flying wing thus forcing me to use APM for long-range flights.

I'm not sure why would this invalidate everyone's config. Just keep the default baud-rate as it currently is, Enable the lower baudrates (<9600) into the next CF release (both for serial and telemetry), and leave GUI as is. 

The source seems to have 9600bps (serial.c, telemetry.c and serial_cli.c) hard coded everywhere (not sure why it isn't comparing against the 1st member of baudrate struct)
```
if (baudRateIndex < BAUD_9600 || baudRateIndex > BAUD_115200) {
                   continue;
```

I've found only one remark for this: 
_LTM is transmit only, and can work at any supported baud rate. It is designed to operate over 2400 baud (9600 in Cleanflight) and does not benefit from higher rates. It is thus usable on soft serial._
Again, I don't seem to get what an author meant, but it seems to support bauds below 9600.

conkerkh, 2016-10-12:
I believe that supported baudrates are written in flash as values represented in enum so instead of using long int  or int, it is possible to save flash and write config just as uint8 which value represents appropriate baudrate. So changing this would change everyone's baudrate to different lower values and would most likely cause confusion because people wouldn't be able to connect at first. 

The easiest way for you is to simply clone cleanflight on your PC and make some changes shouldn't be hard once you get familiar with the code.

hydra, 2016-10-14:
not something we're interested in for Cleanflight right now, users have other more pressing needs.

closed 2016-10-14 by hydra