cleanflight/cleanflight #2419: cli.md is broken

gaelj, 2016-09-21:


tigert, 2016-09-21:
Looks like a conflict pushed by accident.

Git blame shows a commit 8f8428a but since I did not work on that file, I don't know what the correct content should be, so I'll ping @AustinSaintAubin whose commit it was.

AustinSaintAubin, 2016-09-22:
These conflicts where indeed pushed by accident.
Likely due to my inexperience with git and only noticed when [FrSky FLVSS Individual Cell Voltages Telemetry #1927](https://github.com/cleanflight/cleanflight/pull/1927) poll request was accepted.

 I am very sorry for the pollution.
What can I do to fix?
Please include any git commands & links to explain.

tigert, 2016-09-23:
I am not that good with git either, so I guess it happens :)

Maybe you can look through the file so that it renders correctly (the table etc) and that the content is correct, and just edit out the conflict marker lines? I am not sure if there is a more "correct" way to do that.

hydra, 2016-09-25:
fix merged, thanks

closed 2016-09-25 by hydra