cleanflight/cleanflight #2422: Receiver Signal Input Location

johnrusscher, 2016-09-24:
I am using a Naze32 v5 board with cleanflight 1.13.0.  When soldering my receivers signal wire onto pad 4 on the Naze32 board the pad lifted. Is is possible to solder the signal wire to a different pad(maybe 3) and change the signal input location in cleanflight? 

Thanks,
John

jflyper, 2016-09-25:
No.

I'm ready at RCG thread if you have further questions.
[http://www.rcgroups.com/forums/showthread.php?t=2249574](http://www.rcgroups.com/forums/showthread.php?t=2249574)

hydra, 2016-09-25:
@jflyper thanks for helping out with support issues.  @johnrusscher please search before posting, this has been a common question and repeatedly answered.

closed 2016-09-25 by hydra

Dr-Drone, 2016-09-25:
@hydra Sorry for posting it in here. 
@johnrusscher here is the link: http://s21.postimg.org/khia2vwh3/DSC4807_acronaze32_1600.jpg
just solder in your ppm wire there and glue it in place. be careful not to over heat the little resistor.