cleanflight/cleanflight #2426: 3 Foxeer F303 boards flashed properly not working suddenly

daddyvicken, 2016-09-25:
I went through three F303 boards.  I flashed with SPRacing firmware.  Everything was fine.  I was configuring.  I would disconnect connect everything is fine.  By day two or three cannot connect anymore to CF.  I get this message.

2016-09-25 @ 1:04:07 AM -- No configuration received within 10 seconds, communication failed

Help

jflyper, 2016-09-25:
Flash back all three back to the version you were using prior to this update.

Keep No.1 that way
Reflash No.2 to the version you had problem with, power it on/off, connect/disconnect
Reflash No.3 to the version you had problem with, don't power it up

Leave them alone for day or two, then try them one by one.

hydra, 2016-09-25:
sounds like something for foxeer tech support to handle.   Please see the contributing guidelines before opening support issues.

closed 2016-09-25 by hydra

Dr-Drone, 2016-09-25:
First thing, Check your Baud Rate it should be 115200.
No luck then try opening up your device manager and see if your board pops up there. If it doesn't then it should likely be your USB cable, try changing the cable. If that doesn't work then try plugging it in another computer. Or if you have an android phone, try downloading the android app for Cleanflight (EZ-GUI ground station) and plug it in.