cleanflight/cleanflight #2431: BICOPTER, 2 motors works, 2 servos didnot

duypham11, 2016-09-27:
I selected BICopter for my project but servos didnot work, I tried to use smix but it didnot work.
The motor works well, but servo didnot, please have look.
I searched around and saw that there are many question for this kind of issue, but I have not got any replied.

Thanks,
-Duy

hydra, 2016-10-14:
config dump please.

nice to see someone actually wanting to build a bi-copter.  I don't think many people do so it'll be hard to get it tested and working.

can you give details of the frame you're using too?  photos?

hydra, 2016-10-20:
@duypham11 config dump, photos etc please.  if you want this to be looked at you need to help us to help you.

duypham11, 2016-10-20:
I am sorry, I am on vacation so I could not provide those information. I will do that when I am back. One more thing, the servo on rc output4 works, only output4 can make servo works and respond correctly with changes from board.

closed 2017-02-25 by hydra