cleanflight/cleanflight #2434: voltage sensor fluctuation on Lux target, identical to #1817

vzachari, 2016-10-01:
I posted this on #1817 but it's been closed for a long time and burred so I am starting a new issue. Apologies if this is not the right way to go about this.

Issue:

I am getting VBAT fluctuations within +-0.5V with the Lux target, very similar to what initially reported on #1817. This is on CF 1.13.0. The fluctuations are unaffected by clock speed or load (ex. turning softserial on and having all UARTs communicating with accessories). I asked around on the Lux flight controller section on RCGROUPS and some people reported the same behavior. Happy to do any testing if it will help,

hydra, 2016-10-13:
any chance you can look though the bf adc/battery code and see what changed?  what happens if you use an old BF release?  you can use git bisect to track down the commit that fixed it for you.  Once you narrow it down a little porting the changes over should be ok but we need to know what to change first.

vzachari, 2016-10-18:
I'll have a go once I get some free time from work. Most likely this weekend.

hydra, 2016-10-30:
any progress?

closed 2017-02-25 by hydra