cleanflight/cleanflight #2442: Will not find configuration???

Moturbo05, 2016-10-11:
Hello, I am trying desperately to fly my Assassin 180 v2 with the Naze 32 and it will do nothing, when I connect it to cleanflight it opens port then closes port and says no configuration found in 10 seconds and that's all it will do? Anyone have a clue what to do? Thanks.

andre92, 2016-10-11:
Naze has RX TX pins for OSD shared with USB and only one will work at a time. If you have an OSD or anything connected to RX TX pins, unplug and try again. If it works now, stop reading here.
If that doesn't work, you may need to reflash the board using bootloader mode, that should always work. (bootloader requires Zadig driver)

tigert, 2016-10-11:
The github issues are not a support forum, but rather meant for developers to track bug reports and for planning new features.

Please see the rcgroups thread about the Assassin, which likely contains information related to this: http://www.rcgroups.com/forums/showthread.php?t=2579563 - if the question is not answered in the preceding discussion already there, that would be the best place to ask.

SteveAmor, 2016-10-11:
@Moturbo05 please can you close this issue.

closed 2016-10-11 by Moturbo05