cleanflight/cleanflight #2443: Autonomous flying

liutairan, 2016-10-12:
Is there any intention to have autonomous flying or way point flying functions recently? Or does it already exist? Thanks.

digitalentity, 2016-10-12:
Please have a look at INAV project:
https://github.com/iNavFlight/inav
http://www.rcgroups.com/forums/showthread.php?t=2495732

liutairan, 2016-10-12:
Well, thanks. But the INAV just looks so similar to the cleanflight. I can see there are some changes, but any big differences?

digitalentity, 2016-10-12:
@liutairan what did you expect? Pixhawk or Wookong :smile:? You asked for navigation support in Cleanflight - INAV **is** a fork of Cleanflight - no surprise it looks similar. 

But appearances can be deceiving. INAV has completely new navigation core and very different flight control logic.

liutairan, 2016-10-12:
@digitalentity Yeah, I see. But as I wrote in my original post and the subject, what I am looking for is autonomous flight. Cleanflight and INAV have plenty attractive features, and I was asking if there is any intention to have autonomous flying features like Pixhawk or Wookong in the future. Anyway, thank you. I will read the wiki for INAV carefully. Thanks.

digitalentity, 2016-10-13:
Please define "autonomous flight". INAV already has support of waypoint
flight.

hydra, 2016-10-20:
Closing, see INAV for now.

closed 2016-10-20 by hydra