cleanflight/cleanflight #2451: No USB port on the drop down menu...

drone-mechanik, 2016-10-15:
I have the latest version Cleanflight (1.2.3) and Chrome (54.0.2840.59 - 64bit) installed on my iMac running the latest Mac OS X Sierra (10.12). I have installed the CP210x USB to UART Bridge VCP Drivers. I also have tried to force the serial connection using a command line in the terminal. None of the above helped... And I am using a proper data USB cable and not a charging cable.

I've googled around a bit but can't find a solution to my problem? 

If everything is working normally, does the USB port show up in the drop down menu even when there isn't a Flight Controller connected? Or there has to be something connected and powered up in order for it to show in the option? 
I have of course had my FC powered up and connected to my iMac when I was trying to get it to communicate with Cleanflight.

Please help!!! Thanks in advance for all your input!

* I'm trying to connect my TBS Vendetta to Cleanflight, with the Powercube, the FC is Colibri.

rb1205, 2016-10-17:
This is the bug tracking page, not a support service. Even if it was, this is about the firmware, the configurator software (cleanflight-configurator) has its own project page. Head over to some forums like rcgroups and ask for help there.

closed 2016-10-17 by martinbudden

drone-mechanik, 2016-10-18:
sorry I’m all new to this, i will go check at regroups for help thank you