cleanflight/cleanflight #2454: Flying Wing Elevon mixing incorrect

BRDM, 2016-10-16:
I am new to github so hope this follows guidelines etc. I have identified an issue with the wing mode in cleanflight when applied to a naze32 acro.

When connecting a naze32 acro to a flying wing, elevon functions 'pitch' and 'roll' are assigned to the wrong channels. I have confirmed the transmitter and wiring setup by loading baseflight onto the board and I could get it to work, however having used cleanflight on my quads I prefer the superior GUI and cleanflight and it seems to be a better software/programming all round.

I am using a Naze32 Acro on a small EPP flying wing, My controller is setup correctly as TAER and my inputs show correctly in the receiver tab, however my inputs are translated incorrectly to the wing, Roll input creates a pitch response from the Elevons and Pitch input creates a roll response. I can correct this by mapping TEAR so my roll and pitch are swapped in the reciever tab but the correction from the naze in stabilised modes is still incorrect.

I have done lots of reading into this, some people claim there are workarounds by mapping certain servos differently or playing with the checkboxes in the servo tab (which have no intuitive response/function).
Baseflights servos tab in the wing mode is superior here for usability and clear and intuitive input.

One user reported that it may be something to do with the fact that the servos are listed as: 
Motor -  servo1 and servo2
Left Elevon - servo3
Right Elevon - servo4

but in the servos tab the servo's start from servo0. They mentioned in may be a 0 to 1 start point code issue.

If this could be fixed it would be much appreciated, it seems that people are now beginning to branch out from quads and proximity flying with powered wings seems to be gaining popularity pretty quickly, therefore having a working and intuitive wing setup for the naze32 would allow cleanflight to be the much easier to use, go to choice for wing flyers.

hydra, 2016-10-20:
This sounds more like a support issue than an actual bug IMHO.

Posting your config and a photo of your wiring would others to help you.   Did you post your problem on forums before posting here?  This isn't the place for support/setups questions.

Sounds like a board orientation issue or other config issue.  I happily fly an EPP wing using an SPRACINGF3MINI and the code for it is the same as the naze32.

BRDM, 2016-10-20:
Thanks Dominic,

I have actually sorted the control issue and it was a setup thing sorry
(reversing one of the servos) but the confusing servos page made it very
hard to identify, can we change this to a idea for cleaning up the servo
interface page, if I had any idea how to code to do it, i would give it a
go. there are alot of functions on the servos page and very little
information as to what they do and how to use them.

I had researched other forums first and there was very little in the way of
information and most people just defaulted to Baseflight for wings, however
I find cleanflight superior when I can get it to work.

Maybe we could change the title to 'Wing configuration - Make servo
interface page more intuitive and easier to use'

Sorry if this has wasted your time, I have just finished the wing today and
stabilising modes appear to be working okay, It will go for it's maiden
tomorrow.

Thanks
Blair

hydra, 2016-10-30:
that's ok, thanks for the feedback!

closed 2016-10-30 by hydra

BRDM, 2016-11-03:
Hi Dominic,

Not sure if you are still monitoring this, but i have a question.

is there anyway to activate airmode and not have the motor spin when using
the wing configuration ie, Motor stop still works, I find when gliding out
of loops and coming in to land I have almost no control over the wing
because my throttle is at zero and therefore all the control inputs are
severly reduced in rate/acro mode. If I could use airmode, stabilisation
would have full command and I would have full control on 0 throttle glides
and landings. I understand that having the motors stop midair on a quad can
be disastrous (happend to me 10 metres over concrete! very expensive crash
that one) however in wings there is no ill effect to having the motor stop
and then restart because the airframe is inherently stable.

Hopefully you can help, I imagine it would be a pretty simple code change.

Cheers
Blair

martinbudden, 2016-11-03:
@BRDM , have you tried iNav - https://github.com/iNavFlight/inav ?

As well as having enhancements to do with navigation, iNav has good support for flying wings. I'm not a flying wing user myself, so I cannot speak first hand, but I know there is active input from wing users to iNav.