cleanflight/cleanflight #2457: NVIC_IRQChannelSubPriority forced to zero.

asabird, 2016-10-18:
Hello,
**NVIC_IRQChannelSubPriority** is always zero after calling **NVIC_Init**(...)
This is because the macro : **NVIC_BUILD_PRIORITY**(base,sub) from the file **nvic.h**  compute the sub part as 0.

This can lead to a miss priority order in some interrupt handlings.

Thanks for all your great job.

ledvinap, 2016-10-18:
@asabird : Can you be more specific / give some example?
When writing `NVIC_BUILD_PRIORITY`, I simply reversed code used in `NVIC_Init`. I hate the code, but I can't think of better way if we want to use the library ..

asabird, 2016-10-19:
I'm working on it.

asabird, 2016-10-19:
I used this code to show that computed subpriority is not the expected value :
`
printf(" NVIC_PRIORITY_SUB(NVIC_BUILD_PRIORITY(1,0)) : %x expected : 0\n", NVIC_PRIORITY_SUB(NVIC_BUILD_PRIORITY(1,0)));

 printf(" NVIC_PRIORITY_SUB(NVIC_BUILD_PRIORITY(1,1)) : %x expected : 1\n", NVIC_PRIORITY_SUB(NVIC_BUILD_PRIORITY(1,1)));

 printf(" NVIC_PRIORITY_SUB(NVIC_BUILD_PRIORITY(1,2)) : %x expected : 2\n", NVIC_PRIORITY_SUB(NVIC_BUILD_PRIORITY(1,2)));

 printf(" NVIC_PRIORITY_SUB(NVIC_BUILD_PRIORITY(1,3)) : %x expected : 3\n", NVIC_PRIORITY_SUB(NVIC_BUILD_PRIORITY(1,3)));
`

asabird, 2016-10-19:
So I suggest to update: NVIC_BUILD_PRIORITY and NVIC_PRIORITY_SUB macros to :

`#define NVIC_BUILD_PRIORITY(base,sub) (((base)<<(4-(7-(NVIC_PRIORITY_GROUPING>>8))))|((sub)&(0x0f>>(7-(NVIC_PRIORITY_GROUPING>>8)))))

#define NVIC_PRIORITY_SUB(prio) ((prio)&(0x0f>>(7-(NVIC_PRIORITY_GROUPING>>8))))
`
Which produce correct results.

ledvinap, 2016-10-19:
@asabird : NVIC_BUILD_PRIORITY must generate correct value for NVIC registers ..

ledvinap, 2016-10-19:
@asabird : `NVIC_PRIORITY_SUB()` was wrong, it was extracting wrong bits. Correct version is here:
```
#define NVIC_PRIORITY_SUB(prio) ((((prio)>>4)&(0x0f>>(7-(NVIC_PRIORITY_GROUPING>>8)))))
```
Sorry, I tested only NVIC_BUILD_PRIORITY before, not NVIC_PRIORITY_SUB. Thanks for spotting this

asabird, 2016-10-19:
Thank you for the fix.

PR #2460 (2016-10-19): Fix NVIC_PRIORITY_SUB to return correct bits
Fixes #2457
@asbird found this bug, thanks.
closed 2016-10-19 by asabird