cleanflight/cleanflight #2458: USB connection freezes on SPF3(1.13) when powering OSD (1.14)

Enlightx, 2016-10-18:
i can connect to the SPF3 via USB perect. as soon as i add 4S power to the OSD/PDB usb connection to the SPF3 freezes . com port is still present on windows but refuses to talk and connection.

i didnt seem to have this issue when using OSD (1.13)

hydra, 2016-10-20:
you will always have that issue.  The OSD will enable the bus-switch on boot which ties the UART1 on the SPRACINGF3 to the OSD.  UART1 is shared between the stacking pins and the USB socket on the SPRACINGF3.  The SPRACINGF3EVO does not have this restriction.  It's by design.

Basically, if you configure one board at a time you'll be fine.

closed 2016-10-20 by hydra

Enlightx, 2016-10-20:
So the evo doesn't have this issue?

Also with the spf3 just a pain when calibrating rsc etc etc and testing motors. I take it i could reroute the port to another port to solve issue ?

Closed #2458<https://github.com/cleanflight/cleanflight/issues/2458>.

-