cleanflight/cleanflight #2461: Possibility to make the flight controller run as a linux exe?

mkschreder, 2016-10-19:
Can we somehow separate all hardware specific stuff and run the flight controller only as an executable? I have done work simulating flight control for ardupilot controller (see video below demonstrating how simulation works) and what is basically necessary is if the flight controller can run as it's own program and interact with the simulator over shared memory. The simulator then simulates the physics and feeds raw sensor readings to the flight controller and the flight controller writes pwm values to the shared memory that the simulator picks up and uses that to apply forces to motors. You can get in touch with me if you are interested in planing how we can best do this. 

https://www.youtube.com/watch?v=d6iGJySEadc

hydra, 2016-10-20:
For a while now there has been demand to compile and run sections of the core CF code on non-FC targets.  You're not the only person that wants this.  If you'd like to propose and start writing a solution to this I'd be interested.  Other targets include would perhaps include Windows DLLs.  However we would need a maintainer for that.  Also there is probably a lot of cleanup that would need to happen first.

If you would like to be the maintainer of that great, if not then it's likely there would be significant code rot in any solution proposed.

martinbudden, 2016-10-21:
@mkschreder , this is interesting.

The Cleanflight code is now reasonably well separated and it should be reasonably straightforward to do what you want. What is required:

1. Write a gyro device driver that gets its input from shared memory as you describe.
2. Write a receiver module that gets its input from shared memory
3. Read the motor outputs in the main process loop and write the values to shared memory.

martinbudden, 2016-10-21:
@hydra ,

this is probably more useful than you think. If there was a minimum implementation as above, it could be used by the test code. So the test code could set up some gyro and RC inputs, run the pid loop and check that the motor outputs were reasonable.

digitalentity, 2016-10-25:
This will also make SITL tests much easier, i.e. for testing and debugging
navigation code.

ledvinap, 2016-10-25:
IMO almost everything is in place.

@martinbudden: `millis()`, `micros()` needs to be replaced with simulation time too.

My personal experience is that using queue (or similar synchronization mechanism) instead of shared memory will make code simpler and less error-prone ...

How hard would be implementing CF in ROS framework?

martinbudden, 2016-10-25:
I think I'll look at doing this for iNav - since there is a real use case for it, debugging the navigation code. Won't be able to do anything on this until next week though.

mkschreder, 2016-10-25:
We need to have even more separation between modules. Ideally each
subdirectory should be able to compile into a standalone library and then
linked into a final executable. Today there is still far too much stm32
stuff that gets included into global scope. Need to make sure stm32 stuff
never gets included into headers. It should all be completely private to
stm32 related drivers.

My experience is that using shared memory for this is the most efficient
approach because delays matter (very noticible when using high speed
physics simulation and running everything over udp).

jflyper, 2016-10-26:
I'm extremely interested in SITL, too.

hydra, 2016-10-30:
:smile:  what you have all mentioned are great use cases.  the testing one being the most useful one to the core cleanflight project.

closed 2017-02-25 by hydra

thias15, 2017-03-07:
@mkschreder Did you ever figure out how to run betaflight as a Linux process? I'm interested to integrate betaflight with a flight simulator.

sudo-robot-destroy, 2017-05-23:
I'm very interested in this also. Although, I'm not wanting to use a physics simulator, I want run recorded blackbox data through the PID controller on my computer for dynamic modeling purposes.

ledvinap, 2017-05-23:
SITL works ... getting gazebo without Ubuntu is hard.
It is possible to use socat + com0com to connect to configurator and fly around.

@sudo-robot-destroy : Replaying blackbox data won't almost certainly work - simulated system would be open-loop, it's behavior will be different from closed loop and results unusable.

sudo-robot-destroy, 2017-05-23:
@ledvinap : I'm playing with a pet project where I'm trying to fit a dynamic model to blackbox data. We talked about it a little a while back [#1998](https://github.com/betaflight/betaflight/issues/1998), I realize it probably won't work but I'm fascinated by the idea.

I would feed the recorded gyro data into the PID controller code which outputs motor commands. I'd then run the motor commands through a differential equation to calculate the vehicle's response. I'd compare the calculated vehicle response to the real blackbox gyro data and run an optimization of the parameters in the differential equation to hopefully make the two match. The result would be a dynamic model of the quad that could be used for automated offline PID tuning. 

What did you mean by

> getting gazebo without Ubuntu is hard

Why can't Ubuntu be used?

ledvinap, 2017-05-23:
Isn't it equivalent to using logged motor commands? 

I don't use Linux on workstation and run Debian on servers. And resolving gazebo dependencies is a bit of nightmare.

sudo-robot-destroy, 2017-05-23:
Oh I see, so Ubuntu can be use if I wanted to. 

Yes, it is equivalent to using logged motor commands during the dynamic model fitting part. But once I generate an accurate model (if I can), then I would like to have the ability to simulate the PID controller on my computer to run optimization of the PIDs. I would be extrapolating the model to estimate how the vehicle would move using different PIDs. Does that make sense? 

I'd be using the dynamic model to test PIDs rather than flying - that way I can run optimization software that can run through thousands of different PID combinations quickly and compare the estimated vehicle response to the desired response.

mkschreder, 2017-05-23:
You can not really simulate the pids effectively and expect the same
results on the real thing. A more sensible approach is to measure system
step response, measure deadtime between step and start of response and
measure ratio between input command and corresponding system response. Then
you can calculate the pid values that will give excellent results for
controlling the system. For an explanation you can read the controlguru
website http://controlguru.com

sudo-robot-destroy, 2017-05-23:
Controlguru is great! That's where I got the idea- in the sections where they talk about fitting the model using commercial software on data from a doublet test. They're doing essentially what I'm talking about, maybe I'm just bad at describing it.

ohshout, 2017-07-25:
I followed the tutorial here: https://github.com/betaflight/betaflight/blob/master/src/main/target/SITL/README.md

But getting loads of FLASH errors when launching the ELF file. Then it just hangs. Any idea why? Thanks!

The output:
czq@u:betaflight$ ./obj/main/betaflight_SITL.elf 
[system]Init...
[FLASH_Unlock]size = 1, 1
init PwnOut UDP link...0
start UDP server...0
[FLASH_Unlock] eepromFd != NULL
[FLASH_ProgramWord]0x0x63b226 = 4953be9f
[FLASH_ProgramWord]Out of Range! 0x0x63b22a
.....
[FLASH_ProgramWord]Out of Range! 0x0x63b88a
[FLASH_ProgramWord]Out of Range! 0x0x63b88e
[FLASH_ProgramWord]Out of Range! 0x0x63b892
[FLASH_ProgramWord]Out of Range! 0x0x63b896
[FLASH_ProgramWord]Out of Range! 0x0x63b89a
[FLASH_ProgramWord]Out of Range! 0x0x63b89e
.....
[failureMode]!!! 5

ledvinap, 2017-07-25:
You can look at https://github.com/ledvinap/cleanflight/tree/work-SITL. I did change eeprom handling a bit (less linker magic)

I'm cleaning up that branch, but that mainly refactoring ...

ohshout, 2017-07-25:
@ledvinap Thanks for your reply! Can I assume your cleanflight repo currently has the most polished Linux/SITL support that I can find online? Is there any other repo I should take a look? 

Here's what I want to do: I'm trying to 'abuse' the SITL target to get a quick linux port of cleanflight on the Intel Aero board. I've got gyro, acc and mag drivers working in separate Linux processes and cleanflight built as SITL target. But I'm struggling getting raw mag readings through over localhost UDP and then pumping them into the internal Mahony alg, and getting SITL talking to my cleanflight configuration on my separate desktop over TCP.

Here's the original issue I created yesterday: https://github.com/cleanflight/cleanflight/issues/2903

Again, Thanks!!

ledvinap, 2017-07-25:
@ohshout: I don't know .. I just took @cs8425 and played a bit with it ...
But it does work for me, only problem is that it needs some refactoring, there was quite long 'hack it to get it working' phase ..

ledvinap, 2017-07-25:
@ohshout : My branch is mainly about serial port code, connection to gazebo is almost unmodifie

ohshout, 2017-07-25:
@ledvinap All right, thanks a lot! I'll test it out and see it goes.

ohshout, 2017-07-25:
@ledvinap It gives me this:

czq@u:main$ ./betaflight_SITL.elf 
[system]Init...
[FLASH_Unlock] loaded 'eeprom.bin', size = 8192 / 8192
fdmInit()
init PwnOut UDP link...0
start UDP server...0
[led]Init...
[timer]Init...
UART1: can't execute UART handler './uart' : No such file or directory
fdmFd read EOF
Child 11193 terminated

Then it hangs. I have both Gazebo and betaflight configurator open. Below is a little bit system info if that helps:
Linux u 4.4.0-83-generic #106-Ubuntu SMP Mon Jun 26 17:54:43 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux

ledvinap, 2017-07-25:
you need this script in path and executable (it works with SBUS received on FTDI usb-serial, pin inversion has to be enabled with FTProg):

``` bash
#!/bin/bash

>&2 echo "UART $@"

BASE=$(dirname $(readlink $0))

port=$2

case $port in
    1)
        while true; do  socat -d -d - tcp-listen:5761,reuseaddr,nodelay; sleep 1; done
        ;;
    2)
        setserial /dev/ttyUSB0 spd_cust divisor 240 low_latency
        while true; do  socat -d -d - /dev/ttyUSB0,raw,echo=0,cs8,clocal=1,b38400,cstopb=1,parenb=1,parodd=0 ; sleep 1; done
        ;;
    *)
        >&2 echo Port $port is not defined
        ;;
esac
```

ledvinap, 2017-07-25:
You can put about anything in `case`, it will be connected (stdin/stdout) to given serial port

sudo-robot-destroy, 2017-07-26:
I'm having the same issue as ohshout when trying to run SITL: 

```
[FLASH_ProgramWord]Out of Range! 0x0x63b89e
.....
[failureMode]!!! 5
```
Does the script above address that? If so, I'm not sure how to implement it. How does it get called? Or am I suppose to run it manually before starting the .elf?

ohshout, 2017-07-26:
@sudo-robot-destroy I got it working using this branch: https://github.com/cs8425/cleanflight/tree/bf-SITL-fix
I think @ledvinap 's uart script is for forwarding TCP stream to serial port, which I don't need. But I believe @ledvinap 's branch has more polished code due to his refactoring. 

On another note, it seems strange to me that all the fixes related to EEPROM in bf-SITL-fix branch has already been merged into the cleanflight master branch (the master branch even has a few more fixes to the SITL target), but the master fails to work and bf-SITL-fix works. 

I couldn't find why... Any hints from any one?

cs8425, 2017-07-26:
I found the EEPROM problem at CF.
https://github.com/cleanflight/cleanflight/blob/master/src/main/target/SITL/target.c#L467-L468
```c
uint8_t __config_start;
uint8_t __config_end;
```
should be:
```c
extern uint8_t __config_start;
extern uint8_t __config_end;
```
I will create a PR to fix this.

ohshout, 2017-07-26:
@cs8425 well spotted! It does work now! Thanks a lot!

ledvinap, 2017-07-26:
@cs8425 : I did remove linker dependency in my branch ... it is replaced by macros + normal array ..

cs8425, 2017-07-26:
@ledvinap I merge it and create PR cleanflight/pull/2905, betaflight/pull/3633.

tcmichals, 2017-10-05:
I've been working on leveraging the SITL base code into a working Linux FC via SPI/I2C and serial.  One of the issues is having a scheduler use a timer list, socket handlers, serial handlers, Digital input, etc with out threading each one.   Currently using libevent for this, so, it should be portable to windows.  currently the basics are working (MPU9250 via SPI, I2C MS5611, working on serial port for GPS).

sudo-robot-destroy, 2017-10-24:
I get the following when I try to run betaflight_SITL.elf, is there a fix for this? 

[system]Init...
[FLASH_Unlock] loaded 'eeprom.bin', size = 32768 / 32768
init PwnOut UDP link...0
start UDP server...0
Segmentation fault (core dumped)

sudo-robot-destroy, 2017-10-29:
I'm having some trouble using the SITL. I'm able to get it and gazebo running following the directions, but I can't figure out how to connect to the SITL programmatically in order to use it. I've been trying to connect using [pyMultiWii](https://github.com/alduxvm/pyMultiWii). After a little tweaking, I think I'm successfully connecting to tcp://127.0.0.1:5761 but the simulation doesn't respond to any MSP commands or requests. I'm not sure if betaflight can just accept MSP commands like I'm trying to do, so that might be the problem. 

Does anyone have experience they can share interfacing with the SITL using a python library? 
Does MSP work the way I'm thinking it should? (i.e. I'm thinking I should be able to just connect and send MSP_SET_RAW_RC commands and the quadrotor will takeoff in gazebo). 
Are there other interfaces I should try? Does MAVLink work? If so, could I use the [DroneKit](http://python.dronekit.io/) library?

Any help would be appreciated, I'm in over my head but if I can just get to the point where I can send RC commands using Python somehow I think I'll be good. Thanks.

cs8425, 2017-11-01:
@sudo-robot-destroy 
Segmentation fault should be fix after [betaflight@6f4bd2a](https://github.com/betaflight/betaflight/commit/6f4bd2ac06d699bf9f582855da039dc2bb175b20).
cleanflight may need some time to merge.

MAVLink protocol will not work for control, only for some status report.
MSP protocol will work for control and status report.
Send `MSP_SET_RAW_RC` will work as you thought.
As I know, there are no python library for MSP protocol.
I only wrote node.js/js code for my personal testing.
Maybe you can use as references:
https://github.com/cs8425/msp-controller
https://gist.github.com/cs8425/51893e2f90812aa3831558503597fa1a

ledvinap, 2017-11-01:
I was able to control CF using SBUS receiver + FTDI chip (with pin inversion enabled). I need some time to finish it (mainly refactoring) https://github.com/ledvinap/cleanflight/commits/work-SITL

sudo-robot-destroy, 2017-12-13:
Sorry for being a necromancer here, but I'm having trouble finding out how the .elf file communicates with gazebo. I'd like to replace gazebo with my own simple simulator but I can't find out the interface between the two processes. Anyone have any ideas?

ledvinap, 2017-12-13:
@sudo-robot-destroy : UDP packets are exchanged:
https://github.com/ledvinap/cleanflight/blob/work-SITL/src/main/target/SITL/target.c#L78
https://github.com/ledvinap/cleanflight/blob/work-SITL/src/main/target/SITL/udplink.h#L31

tcmichals, 2017-12-19:
FYI, I'm trying to enhance the Linux port also:
- Used libEvent to standardize on event framework based on epoll and timerfd (Updated the scheduler to use libevent)
- created a protocol to allow forwarding PWM/dshot/LCD messages to external board via UDP or TCP
Working on fixing CLI and MSP to use a message queue.  This allow, CLI/MSP to be access via multiple transports i.e. linux command line, TCP and serial.    

https://github.com/tcmichals/cleanflight

wil3, 2017-12-29:
@sudo-robot-destroy I've modified pyMultiWii to connect over TCP, best programic solutions I've come across, https://github.com/wil3/pyMultiWii/tree/feature-tcp. Usable but still needs work before merging upstream.

chrisalbertson, 2018-07-26:
about " how the .elf file communicates with gazebo. ".     I Hate to insult people by using an over simple explanation.    Sorry if it is...

The answer is that igazebo communicates the same way all processes in ROS communicate.  Using ROS' publish and subscribe model.    You have to write yu own "yellow world" ROS app to understand this.   Basically gazebo subscribe to a set of message types and registers a callback for each one.  Then waits.

I would NOT replace gazebo.  It is a classic beginner mistake  to say "I don't understand this so I will write another one myself" 

I think I might use a different method to get ROS/Flight Controller interfaced.    I will have the Linux machine make itself "look" like a receiver.   This might be better because then the Linux software is not tied to Beta Flight but can talk to any SBUS device.

My application id antonymous robotics.  For example I want a drone that will follow a person or a dog and maintain a given angle so that as the person rotate the drone will orbit around and at the same time the drone watches for and avoid obstacles.    This is kind of like a simplified self-driving car and would needs a LIDAR and a camera

ROS can do localization,  navigation and path planning and near state of the art levels

I doubt you can do fast control loops under Linux.   Line will multitask and  your process can halt at un predictable times for several milliseconds.    There are a few versions of Linux that are modified for "hard" real time.  You would need this for a flight controller.   Bt RT Linux kills the performance of non-RT tasks.     MUCH better I think to runs the auto-pilotof and cheap FPV type controller.

That is what I do on my ground based robots.  Typically the hard RT tasks get do on an ARM Cortex-M and the higher level planning on the Linux side.