cleanflight/cleanflight #2464: suggestion for avoiding ifdefs

mkschreder, 2016-10-25:
To avoid ifdefs in the code you can change all ifdefs that are used in target.h and give them values 1 or 0. Also you combine this with -fdata-sections, -ffunction-sections and -Wl,--gc-sections (which is already enabled). 

So instead of: 

`#define USE_SERVOS`

Use: 

```
#define USE_SERVOS 1
or 
#define USE_SERVOS 0
```

Then in the other code, instead of: 
```
#ifdef USE_SERVOS
...
#endif 
```

Use: 

```
if(USE_SERVOS){
...
}
```

Any decent compiler will completely eliminate the code in that if statement because it always evaluates to false. Together with function and data section elimination you also eliminate all data that may be referenced by these blocks. The major upside is that compiler can check that code for syntax errors which is not possible when ifdefs are used.

hydra, 2016-10-30:
yeah, that's a pretty good idea. and i like the plus-side too.

@ledvinap @martinbudden @borisbstyle thoughts?

martinbudden, 2016-10-30:
I believe this is the recommended practice for Linux development. I think idea is good, but any changes in this direction should wait until after current release.

Note this will be a "noisy" change. We should avoid combining cleanup PRs with additional features.

ledvinap, 2016-10-30:
I'm used to old `#ifdef` way, but this sounds reasonable. Just don't overuse it. `--gc-sections` may have some unexpected consequences, but it should be quite easy to fix ...

mkschreder, 2016-10-30:
For reference: https://www.kernel.org/doc/Documentation/CodingStyle

closed 2017-02-25 by hydra