cleanflight/cleanflight #2466: Missing backup and restore button in cleanflight setup

Nokistar, 2016-10-26:
I Guys,

I need some helpto solve a wired problem with cleanflight.

My backup and restore button has disappeared.

You can find a picture attached.

If anybody has seen it could you please tell me where i can find them ?

Thank's a lot ! 
[image]

aln37, 2016-10-27:
yes I have thre same exact problem and i accidently hit reset and now i dont know what to do is there any way to do a restore from the command line in the CLI? please help

hydra, 2016-10-30:
it's not weird, it's documented in the changelog.  please search before opening new issues.

closed 2016-10-30 by hydra

9145916889, 2017-01-29:
Question I have the old backups on my quads thru the clmeanflight 1.1 backup button how can I use it to restore on 1.24 thanks I did the dump and CLI restore but the MODES did not come thru properly?
any help would be appreciated

aln37, 2017-01-29:
You have to go into the settings of Google then go to extensions enable
developer mode once you've done that you need to unpack a bin file of a
previous version of cleanflight you can have both of them installed
simultaneously that older version 1.2 point three still has the backup and
you can use that you will just have to different icons for cleanflight
rather than one that's pretty much what I had to do it's still a little bit
Goofy and buggy that when it comes to transferring stuff back over I found
that the best thing to do is just wipe it completely out and start over
some get screwy especially with the new platform where you can change the
loop times be careful with that if you noticed your processor running at
400% never ever try to fly it like that it does somersaults and crazy crazy
stuff you gotta basically to get that to stop you have to wipe it out as in
hit reset reset reset then refresh close application open application
unplugged flight controller plug it back in until you see your processor
speed go down to zero and then start over never go over I don't know 1K for
now and with an F3 chip seems to be plenty I've just learned how to start
from scratch and reset everything basically buy memory which is probably
your best bet trying to restore previous backups is actually not very
accurate because certain things don't change if you'll notice your PID
settings don't change usually that's another one you have to go in there
and reset them reset them reset them unplug plug back in start-stop that
kind of stuff to get it to zero out or go back to default so you're better
off learning where your settings that kind of memorizing them because you
going to be doing it over and over trust me just got to learn how to do it
just remember it part of the game because after going through all the hell
of restoring your previous version and then trying to reinstall my backups
certain things don't change of course they're the things that I wanted to
change and they didn't so so much for that you also might want to try
betaflight I kind of like that a little better it kinda depends on what
you're doing but good luck with that sorry this is kind of rambly and very
punctuated I was just kind of speaking off-the-cuff here hopefully it made
sense have fun