cleanflight/cleanflight #2469: Fix the angle calculation

mkschreder, 2016-10-27:
In all versions of cleanflight the imu code calculates pitch and roll angles from DCM vectors as follows: 
```
self->attitude.values.roll = lrintf(atan2_approx(self->rMat[2][1], self->rMat[2][2]) * (1800.0f / M_PIf));
self->attitude.values.pitch = lrintf(((0.5f * M_PIf) - acos_approx(-self->rMat[2][0])) * (1800.0f / M_PIf));
```
This is incorrect and results in huge variations of pitch and roll around the singularity point when nose is pointing down. 

The correct way is like this: 
```
    float px = self->rMat[2][0];
    float py = self->rMat[2][1];
    float pz = self->rMat[2][2];
    float sign_z = (pz > 0) - (pz < 0); 

    self->attitude.values.roll = lrintf(atan2_approx(py, sign_z * sqrtf(pz * pz + 0.01f * px * px)) * (1800.0f / M_PIf));  
    self->attitude.values.pitch = lrintf(atan2_approx(-px, sqrtf(py * py + pz * pz)) * (1800.0f / M_PIf)); 
```

This still results in a singularity but the adverse effects are quite minimal and only similar to when you look down in an fpv game where your camera is euler angle based - the error only effects the heading which can become random as opposed to both pitch and roll as in current cleanflight approach. 

For reference see this application note eq 37 and 38: 
https://www.nxp.com/files/sensors/doc/app_note/AN3461.pdf

ledvinap, 2016-10-28:
I don't see much improvement using this:
equation for pitch is identical (but CF uses different offset)
Equation for roll will be still dominated by sign(z), sign(y) when py and pz are close to zero - there will still be discontinuities, only atan(0,0) is suppressed ..

digitalentity, 2016-10-28:
I don't see much improvement in this either. The proper approach to the attitude discontinuity problem is to use quaternion algebra in self-leveling code and provide euler angles only as a reference for telemetry.

mkschreder, 2016-10-28:
The improvement is visible as pitch approaches the singularity point. I can
see a clear improvement when board is pointing nose down. With the existing
code the angles go crazy. With the new code only yaw is affected. Even
though it doesn't really matter during normal stable flight, it is still
worth noting that current approach is not an optimal approximation.

hydra, 2016-10-30:
i tried the existing code out, noted the row angles 'go crazy' when the board is at 90 degrees.
for fun I tried the alternative code out, above:

snippet:
```
    /* Compute pitch/roll angles */
    float px = rMat[2][0];
    float py = rMat[2][1];
    float pz = rMat[2][2];
    float sign_z = (pz > 0) - (pz < 0);

    attitude.values.roll = lrintf(atan2_approx(py, sign_z * sqrtf(pz * pz + 0.01f * px * px)) * (1800.0f / M_PIf));
    attitude.values.pitch = lrintf(atan2_approx(-px, sqrtf(py * py + pz * pz)) * (1800.0f / M_PIf));

//    attitude.values.roll = lrintf(atan2_approx(rMat[2][1], rMat[2][2]) * (1800.0f / M_PIf));
//    attitude.values.pitch = lrintf(((0.5f * M_PIf) - acos_approx(-rMat[2][0])) * (1800.0f / M_PIf));
    
```

and noted that the roll angle immediately flips between -150 and -26 when the board is rotated on the pitch axis as the pitch angle approaches 90 degrees.

ledvinap, 2016-10-30:
@hydra: It is probably possible to implement desired behavior (bring roll to zero near singularity), but there is no way to avoid singularity using RPY angles. Pitch will jump from -180 to 180, confusing PIDs etc. Attenuating Roll may hide the problem a bit, possibly with unforeseen consequences.

Implementing correct behavior using singularity-free representation is IMO way to go ..

ledvinap, 2016-10-30:
@hydra: 
```
attitude.values.pitch = lrintf(atan2_approx(-px, sqrtf(py * py + pz * pz)) * (1800.0f / M_PIf));
```
 is shifted by 90 deg with respect to CF conventions. 
```
attitude.values.pitch = lrintf(asin_approx(px) * (1800.0f / M_PIf));
``` 
should compute correct value (or `lrintf((atan2_approx(-px, sqrtf(py * py + pz * pz)) -  0.5f * M_PIf )* (1800.0f / M_PIf))`, they are equivalent due to `x^2+y^2+z^2=1`

ledvinap, 2016-10-30:
@hydra: the -150 to -26 jump may be some bug in code ...

mkschreder, 2016-10-30:
Actually the "jump" is a combination of changes in yaw roll and pitch together. Also the board does not really flip in the configurator - it just jerks. This is expected because around the poles you have a circle that gets smaller and smaller until you have a point which you can not really represent with euler angles. The behavior is in line with how a 3d camera behaves when calculations are done using euler angles. I tend to trust that behavior more than the randomness observed with approach that is in use today.

ledvinap, 2016-10-31:
@mkschreder : Well, my 3d imagination is a bit too poor for this. Takes time ...

It makes sense, at least for display purposes ... I would still prefer singularity-free solution, but that will take some time ..

ledvinap, 2016-10-31:
It would be great to implement some simple visualization tool for quad orientation ...

digitalentity, 2016-10-31:
@ledvinap I'm looking into quaternion PID-controller for INAV but it's going to be very tough on CPU and will require some time to implement and debug.

ledvinap, 2016-10-31:
@digitalentity : Fast PIDs can be closed over angular rates (= gyro output), no singularity/problem there.
And difference between two quaternions is quite easy. So on first glance the problem shall reduce to determining desired orientation (but I may be very wrong ... )

Any link to your work?

digitalentity, 2016-10-31:
No work yet, just a bunch of ideas.
The general idea is this.

Given: 
 1. We have an orientation quaternion from IMU
 2. We have desired euler angles from sticks

Process:
 1. Initialise desired orientation quaternion from sticks (not sure how to blend heading here)
 2. Find a rotation that is necesary to transform current orientation into desired orientation (this will yield rotation quaternion)
 3. Decompose rotation into roll/pitch/yaw rates
 4. Feed those rates into rate PID

digitalentity, 2016-10-31:
I'm not very good at quaternion math, so any input is greatly appreciated :smile:

ledvinap, 2016-10-31:
@digitalentity, @mkschreder : 
Speaking of stick command in angle mode - what will happen if axis/angle notation is used?
Axis being in plane of TX, perpendicular to direction of stick deflection and angle representing amount of stick deflection. 
This representation will have singularity at centered stick, but deflection is zero, so it should not cause (big) problems.
It can easily be transformed to quaternion for control purposes. It is also quite easy to visualize mentally.
Also angle can be scaled, allowing more than 180deg deflection (but I'm not sure how heading will blend in)

mkschreder, 2016-10-31:
Petr, a singularity free solution already exists. Search for quaternion
based pid. Basically you calculate axisPID outputs based on a desired delta
quaternion. However we already do somethibg that is very similar to this
using the gyro rate quaternion (a quaternion where you have gyro rates as
xyz and w=0 is basically the delta quaternion that cöeanflight already uses
for integrating the dcm).

mkschreder, 2016-10-31:
I believe px4 uses predominantly quaternion based pid controller. These run
fine on f4 chips. I'm pretty sure that even the f3 will handle it just fine
because of the fpu. Ideally we should not be using euler angles at all for
anything that desires reliability because they are unstable.

Btw axis angle is "basically" like a quaternion except you have cos/sin
values stored in the quaternion instead of axis/angle values. Quaternions
are easier to work with. They are basically intermediate form of axis angle
that is stored in such a way that combining rotations is more efficient.

ledvinap, 2016-10-31:
@mkschreder : In ANGLE (and HORIZON) mode, euler angles (roll/pitch) are used, but mainly as mapping from stick input to desired orientation. ACRO mode does not use auler (or pose) at all, the loop is closed through pilot ... 
For quaternion based controller, stick input to desired orientation is necessary (and then there is HORIZON mode ..). The rest is quite easy.

As for performance - Rate control loop may run at higher frequency if necessary (but error computation is faster that pose estimation; simple quaternion cross product should do).

mkschreder, 2016-10-31:
Petr, actually I think that current imu code is close to perfect for what
CF is designed to do (rate mode racing). For areal photography that is done
predominantly in stabilized mode there are other controllers like ardupilot
and px4 that have better angle estimation. But for what this firmware is
doing it does not really matter. I just noticed that the calculation was
not optimal so I pointed it out. In retrospect it is a really minor thing.
It's kindof more important right now to prepare the code for SITL and clean
it up for next round of major improvements when people will be able to test
their changes racing around in a quake map. :-)

HaukeRa, 2016-10-31:
Quaternion math is not so difficult if you wrapped your head around it once. I think I can help out here or there if a quaternion basedo IMU and controller is wanted in cleanflight

digitalentity, 2016-10-31:
@HaukeRa not sure about Cleanflight, but it's definitely very wanted in INAV https://github.com/iNavFlight/inav/issues/272

hydra, 2016-11-17:
ok guys, i will close this issue as there's nothing to be done right now. @mkschreder is right, we have more important stuff to do right now.  Such as IO changes and F4 support.

closed 2016-11-17 by hydra