cleanflight/cleanflight #2472: OLED Display

bgga, 2016-10-29:
Hi.

After updating firmware to 1.14.0-RC2 on Flip32+ board the OLED display won't work.
On previous firmware display worked properly.

hydra, 2016-10-30:
please try resetting to defaults.  please provide further details of your wiring too.  certainly on a couple of boards i tested with RC2 the OLED display is working fine.  however there have been some changes to the I2C handling recently which were done after RC2.  Please try a new development build from here:

http://cleanflight.memoryleaks.org/builds/

please report your findings.

bgga, 2016-10-30:
Display is connected to i2c outputs on the board. After flashing last stable build (1.13.0) display starts normaly and worsk fine.
Under that link i found build from last month only.

hydra, 2016-11-02:
ahh, it's not that, the DISPLAY feature was disabled on the NAZE target due to code size reasons.  I'll check again and see if it can fit.

hydra, 2016-11-02:
nope, it doesn't fit.  Basically there's too much code.  If you need I2C display support on your board then you will have to compile cleanflight yourself and disable some of the things you don't need so it will fit.

This is why we need a modular system so users can choose what they need and what they don't.

@borisbstyle @martinbudden @ledvinap @blckmn 😄

closed 2016-11-02 by hydra

jflyper, 2016-11-11:
Is there any place the modular system is being discussed on?

martinbudden, 2016-11-11:
As far as I know, there is no ongoing discussion about a modular system.