cleanflight/cleanflight #2474: Modify the CF parameter with Frsky LUA and the S.Port Telemetry

bbc25185, 2016-10-31:
Frsky has developed the new receiver S6R...
This receiver can be configured using a FRSKY Taranis radio and the S6R LUA code without using a PC. (To do this, the LUA use the telemetry link).
Since CF already supports the  FrSky S.PORT telemetry,  I think it is quite simple to add support for changing parameters using a specially prepared LUA code to be run on TARANIS, no ???

jflyper, 2016-11-02:
This might be related: betaflight/betaflight#1438

raphaelcoeffic, 2016-11-02:
The original issue with feature discussion is betaflight/betaflight#1311.

raphaelcoeffic, 2016-11-11:
Please note that there is now a testable version for betaflight. Please let me know if I should work on a port for cleanflight.

hydra, 2016-11-17:
yes please!

raphaelcoeffic, 2016-11-18:
@hydra where should I place the LUA script to test the feature? In BF, this is in the test dir, but I think it would be better somewhere else, perhaps even outside the project.

hydra, 2016-11-18:
@raphaelcoeffic create 'support\opentx' and place it there.

floaledm, 2017-01-19:
Great work! Any reason why this hasn't been merged yet?

closed 2017-02-25 by hydra