cleanflight/cleanflight #2476: Unit tests only cover around 1.3% of the code

mkschreder, 2016-11-03:
Are you guys using coverage when writing unit tests? I know unit tests are good and stuff but without doing coverage analysis to write better unit tests, the unit tests will never be adequately exhaustive. 

Running lcov shows only a handful of files being tested in src/main folder and testing is rarely at 100%. 
```
Reading tracefile coverage.info
                                   |Lines       |Functions  |Branches    
Filename                           |Rate     Num|Rate    Num|Rate     Num
=========================================================================
[/data/software/tilt-1.10.0/src/main/]
common/encoding.c                  | 100%      5| 100%     2|    -      0
common/filter.c                    |32.7%     49|40.0%     5|    -      0
common/maths.c                     |59.6%    146|52.2%    23|    -      0
common/streambuf.c                 |71.7%     46|76.9%    13|    -      0
common/typeconversion.c            |25.5%    145|30.0%    10|    -      0
common/utils.h                     | 0.0%      1| 0.0%     1|    -      0
config/config.c                    |74.1%     81|70.0%    10|    -      0
config/config.h                    |84.6%     39|86.0%    43|    -      0
config/config_eeprom.c             |93.2%     88|87.5%     8|    -      0
config/config_streamer.c           |88.4%     43|85.7%     7|    -      0
config/config_system.h             | 100%      1| 100%     1|    -      0
config/feature.c                   |61.1%     18|57.1%     7|    -      0
config/feature.h                   | 100%      1| 100%     1|    -      0
config/parameter_group.c           |80.4%     51|77.8%     9|    -      0
config/parameter_group.h           | 100%      4| 100%     4|    -      0
config/profile.c                   |85.7%      7| 100%     2|    -      0
config/profile.h                   | 100%      1| 100%     1|    -      0
config/vars.c                      | 100%     55| 100%    12|    -      0
drivers/barometer_bmp085.c         |36.0%    100|27.3%    11|    -      0
drivers/barometer_bmp280.c         |49.2%     59|37.5%     8|    -      0
drivers/barometer_ms5611.c         |49.4%     83|20.0%    10|    -      0
drivers/buf_writer.c               | 0.0%     17| 0.0%     3|    -      0
drivers/light_ws2811strip.c        |12.2%     49| 9.1%    11|    -      0
drivers/sonar_hcsr04.c             | 0.0%     11| 0.0%     3|    -      0
flight/altitudehold.c              |23.2%     99|33.3%     9|    -      0
flight/anglerate.c                 |46.7%    212|64.7%    17|    -      0
flight/failsafe.c                  |88.3%    137|87.5%    16|    -      0
flight/gps_conversion.c            |95.5%     22| 100%     1|    -      0
flight/imu.c                       |71.6%    232|58.3%    24|    -      0
flight/mixer.c                     |32.2%    152|31.2%    16|    -      0
flight/servos.c                    |56.1%    189|66.7%    15|    -      0
io/ledstrip.c                      |28.6%    350|29.2%    24|    -      0
io/ledstrip.h                      | 100%      5| 100%     3|    -      0
io/msp.c                           |51.3%    754|80.0%    15|    -      0
io/rc_adjustments.c                |57.3%    157|66.7%     9|    -      0
io/rc_controls.c                   |16.9%    142|45.5%    11|    -      0
io/rc_controls.h                   | 100%      1| 100%     1|    -      0
io/serial.c                        |13.6%    147|13.6%    22|    -      0
io/serial_msp.c                    |63.4%    112|54.5%    11|    -      0
rx/rx.c                            |19.3%    192|24.1%    29|    -      0
scheduler.c                        |82.0%    128|85.7%    14|    -      0
sensors/battery.c                  |52.1%     96|33.3%    15|    -      0
sensors/boardalignment.c           |75.9%     58|66.7%     3|    -      0
sensors/sonar.c                    | 0.0%     35| 0.0%     6|    -      0
telemetry/hott.c                   |17.4%    218| 7.1%    28|    -      0
=========================================================================
                             Total:|48.0%   4538|51.8%   494|    -      0
Summary coverage rate:
  lines......: 48.0% (2180 of 4538 lines)
  functions..: 51.8% (256 of 494 functions)
  branches...: no data found

```

There are 165576 lines of code in that folder in total, most of which is not being unit tested. (that's including generated mavlink code. If I exclude that we are at 62282 lines that potentially need to be unit tested. Current unit tests in that case still only cover 3.5% of the code)

unitware, 2016-11-06:
Hi, I was trying to find the documentation on how to get this report, could not find it. I guess that's a good place to start? Building on a Mac I have got gcov and I get lots of output from that, but not as good looking as the table above.

I've been using other tools (lcov) in other projects to get a nice report but it seems like you got something in place already?

mkschreder, 2016-11-06:
Yeah I have it already added in ninjaflight. When you run "make test" it
will output the report once all tests have passed (my repo is a major wip
though so some tests may actually fail when you build it). As I do more
refactoring of the source I also update the tests accordingly.

You can get the coverage table using lcov like this:
https://github.com/mkschreder/ninjaflight/blob/master/Makefile#L928-L931

Also I have updated my travis script so that it uploads the coverage into
into coveralls and gives nice graphs like this:
https://coveralls.io/github/mkschreder/ninjaflight You can also convert
lcov info file directly into html using genhtml utility on linux and use
your browser to view it.

I'm actually refactoring quite a lot of things including the imu and mixer
and I'm also doing work on adding SITL support (which actually works
already by building only imu and mixer into a shared library and loading it
into a simulator. I just have not written any documentation for it yet). If
you are interested in pulling any of my changes, feel free to do so. Just
be aware that unless it is 100% covered and unit tested, it can never be
guaranteed to work :-)

unitware, 2016-11-07:
Interesting, I have started to copy your solution:
Coverall report: https://coveralls.io/github/unitware/cleanflight
Work branch: https://github.com/unitware/cleanflight/tree/lcov-coverage-report

It works fine on the servers but not on my local builds. I will try to get it working soon and any help is appreciated. There is also a need to document how to set this up on other build architectures (windows, linux) Any help in documenting this is appreciated.

Until then I'll add a check for lcov and ignore those lines if it is not installed.

hydra, 2016-11-17:
@mkschreder I have not had time to look into the coverage reports myself, being quite familiar with the codebase I know that *shedloads* of it, is not covered at all.  

As you know I did a whole bunch of work cleaning up the dependencies and highlighting them in cleanflight and this work has largely been ignored by most contributors, there are only a few like-minded developers who see the value of testing in the Cleanflight/Betaflight/iNav sphere.   Recently I've, clearly, had to shift focus from cleaning code to supporting and maintaining code whilst also battling with all the new features everyone wants.  Not enough time for code cleaning and writing tests even though it's one of the most important thing that the project needs.

I would love to see a PR to update the travis builds scripts to generate coverage reports.

Also, I do not want to see your time going to waste with ninjaflight - I note that there are no stars or follows of the project.  I'd love to see you just work on the cleanflight codebase instead.  Can you make PR's for your changes?

unitware, 2016-11-17:
@hydra have a look at this PR: https://github.com/cleanflight/cleanflight/issues/2487

martinbudden, 2016-11-17:
@mkschreder , I concur with @hydra here. Rather than create a fork that is an even cleaner Cleanflight I think it would be more productive for you to submit PRs to clean up Cleanflight, or add tests. @hydra is very receptive to PRs that involve either cleanup or tests.

mkschreder, 2016-11-17:
@hydra: ninjaflight has totally different focus. I have cleaned up a lot of
the codebase and a lot of it is pretty much portable now - including the
SITL code, IMU and mixer (latest stuff has not been merged into master
branch yet).. It is all rather generic, unit tested, and will work as part
of any project. If you want to take parts of that codebase you are welcome,
but I use a different coding style and I do things very differently in
general. I'm open to collaboration but when it comes to how much code later
ends up in cleanflight is entirely up to you. I have no interest in
conforming to your coding style and to the goals of cleanflight project in
general - so why should I attempt to force you to conform to my way of
doing things by posting pull requests to cleanflight that pretty much go
against your preferred style?

unitware, 2016-11-18:
I think this discussion strayed a bit from the initial subject which we all agree on. 
I suggest that you look at the PR https://github.com/cleanflight/cleanflight/pull/2487 mentioned above that does implement the coverage analysis report and reports it to coverall.io. 

Someone that can log in as github/cleanflight/cleanflight must also create an account on coverall.io. which is easily done and free.

As a side note:
 - Never trust in unittests only, ideally there should be some scenario tests that check that stuff is ptoperly talking to each other apart from the linker.
 - 100% coverage is not the same as working code, the tests must be written well and reviewed.
 - For this reason I think it's dangerous to set some minimal coverage requirement since it does not prove anything, nor compare metrics between projects without proper analysis.
 - 1.3% coverage is not that good :)

mkschreder, 2016-11-18:
I think automatic "mission tests" are going to be pretty good for ensuring
things work as they should. Mission tests are basically automatic flights
that fly inside a simulated environment and have a specific goal - ie to
reach a destination after doing something (PX4 uses these for their code
internally). This can be done deterministically provided that physics
simulation steps are always the same. Ideally I want to automate as much of
the testing as possible. Unit tests are step 1, SITL step 2 and mission
tests are for later. Even manual SITL flights are quite revealing and an
excellent way to test new stuff. However the code needs more work in order
to use all of the flight controller in SITL. This is currently a work in
progress. The biggest benefit of mission tests (and of unit tests in
general) is that you end up writing better code and you can be sure that
given the inputs that you provide, your code will produce the expected
outputs. There is of course good and bad unit testing - unit tests that
just test idle values are quite pointless even though they increase code
coverage. Need to test extreme values and even random values. Coverage will
tell you if you have code that is not tested at all that may contain bugs
that you don't even know about.

+1 for everything Magnus said.

willanth, 2017-02-05:
Is there a style guide for the project that is published, and a guideline for the toolchain you are all using for testing and evaluation?  I have only done a couple years of Comp Sci, and have only done a moderate amount of testing and never on a codebase this large.  If you all would be willing to point the way a bit I am happy to contribute in doing code cleanup and writing unit tests.  Most of my experience is in writing embedded code for standalone applications, nothing ever this sophisticated so there may be aspects that I am unable to work with at first (threading, etc).  I'm a hardware guy mostly, not a software engineer but I'm willing to contribute if there's a way I can be useful.

hydra, 2017-02-05:
@willanth check the 'docs' folder of the project.  also check out the other tests to see how they are done.  the pid controller tests are somewhat more complicated, so look for a test that tests something simple to get you started.

willanth, 2017-02-09:
Thank you, and apologies.  I located the docs folder and drilled down into that a couple days after posting that.  I will review all of that, and start trying to find very simple tests I can write based on the existing ones.  Hopefully I can be of help.

closed 2017-02-25 by hydra