cleanflight/cleanflight #2478: make all_targets fails with v1.14.0

richard-scott, 2016-11-03:
I get the following problem compiling refs/tags/v1.14.0:

```
/usr/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/bin/ld: obj/main/cleanflight_CJMCU.elf section `.data' will not fit in region `FLASH'
/usr/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/bin/ld: region `FLASH' overflowed by 116 bytes
collect2: error: ld returned 1 exit status
Makefile:1024: recipe for target 'obj/main/cleanflight_CJMCU.elf' failed
make[1]: *** [obj/main/cleanflight_CJMCU.elf] Error 1
make[1]: Leaving directory '/var/lib/jenkins/workspace/Cleanflight'
Makefile:954: recipe for target 'CJMCU' failed
make: *** [CJMCU] Error 2
```

youngday, 2016-11-06:
make SPRACINGF3,  make sure your target,  not all is ok now

richard-scott, 2016-11-06:
Are you saying this is fixed now? If so then I'm sorry to say its not,

youngday, 2016-11-11:
cjmcu is atmel 8bit mcu ,you `d better  use new target, stm32f1 or stm32f3

youngday, 2016-11-11:
IN betaflight  forked from cleanflight 

Linking CJMCU
./tools/gcc-arm-none-eabi-5_4-2016q3/bin/arm-none-eabi-size ./obj/main/betaflight_CJMCU.elf
   text	   data	    bss	    dec	    hex	filename
  58092	    784	   8768	  67644	  1083c	./obj/main/betaflight_CJMCU.elf
./tools/gcc-arm-none-eabi-5_4-2016q3/bin/arm-none-eabi-objcopy -O binary obj/main/betaflight_CJMCU.elf obj/betaflight_3.1.0_CJMCU.bin
make[2]: Leaving directory '/home/youngday/stm32/betaflitht/betaflight'
make -j ./obj/betaflight_3.1.0_CJMCU.hex
make[2]: Entering directory '/home/youngday/stm32/betaflitht/betaflight'
./tools/gcc-arm-none-eabi-5_4-2016q3/bin/arm-none-eabi-objcopy -O ihex --set-start 0x8000000 obj/main/betaflight_CJMCU.elf obj/betaflight_3.1.0_CJMCU.hex
make[2]: Leaving directory '/home/youngday/stm32/betaflitht/betaflight'
make[1]: Leaving directory '/home/youngday/stm32/betaflitht/betaflight'
Building CJMCU succeeded.

richard-scott, 2016-11-11:
Still the same in v1.14.1, but overflows by more bytes now!

```
/usr/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/bin/ld: region `FLASH' overflowed by 124 bytes
collect2: error: ld returned 1 exit status
Makefile:1024: recipe for target 'obj/main/cleanflight_CJMCU.elf' failed
```

hydra, 2016-11-17:
this is the cleanflight issue tracker, not the betaflight one.

hydra, 2016-11-17:
@richard-scott it works here using:
```
arm-none-eabi-gcc.exe (GNU Tools for ARM Embedded Processors) 5.4.1 20160609 (release) [ARM/embedded-5-branch revision 237715]
```

it works on travis using:
```
gcc-arm-none-eabi-5_4-2016q2-20160622-linux.tar.bz2
```

if you're using something else YMMV - If you can create some PR's that reduce code size that would be the best solution for everyone.

closed 2016-11-17 by hydra