cleanflight/cleanflight #2479: CPU load in 1.14.0 too high (F3 FC)

lkaino, 2016-11-05:
I rebased Triflight on top of 1.14.0, but I noticed that the CPU load is too high by default.

I then tested with stock CF 1.14, the situation is same there. I'm testing on RCEXPLORERF3 target.

By default the CPU load is around 520%.

```
# tasks
Task list          max/us  avg/us rate/hz maxload avgload     total/ms
 0 -       SYSTEM       4       0       4    0.5%    0.5%         0
 1 -         GYRO     247     209    3257   80.9%   68.5%     87857
 2 -          PID      72      36    1293    9.8%    5.1%      7814
 3 -        ACCEL     217     186     332    7.7%    6.6%      9238
 4 -     ATTITUDE      65      33      48    0.8%    0.6%       276
 5 -           RX      59      28      24    0.6%    0.5%       157
 6 -       SERIAL    1143      10      35    4.5%    0.5%      1440
 7 -      BATTERY       8       0      23    0.5%    0.5%        11
 8 -       BEEPER       6       0      35    0.5%    0.5%         9
```
```
# status
System Uptime: 145 seconds, Voltage: 0 * 0.1V (1S battery - NOT PRESENT), System load: 5.07
CPU Clock=72MHz, GYRO=MPU6000, ACC=MPU6000
PIDd: 994, GYROd: 234, I2C Errors: 4, registry size: 51
```

All settings are at 1.14 defaults:
`gyro_sync = 1
gyro_sample_hz = 4000
gyro_lpf = OFF
pid_process_denom = 2`

GYRO task is taking 209 us. Target cycle time is 250 us. Are the defaults just too demanding?

If I lower `set gyro_sample_hz = 2000`, the CPU load drops to ~120%. PID denom still at 2 (PID task running at 1000 Hz). We can see that gyro 

```
# tasks
Task list          max/us  avg/us rate/hz maxload avgload     total/ms
 0 -       SYSTEM       4       0       9    0.5%    0.5%         2
 1 -         GYRO     247     203    2178   54.2%   44.7%     57597
 2 -          PID      84      34    1300   11.4%    4.9%      7863
 3 -        ACCEL     234     189     806   19.3%   15.7%     26801
 4 -     ATTITUDE      65      34      77    1.0%    0.7%       576
 5 -           RX      59      27      49    0.7%    0.6%       340
 6 -       SERIAL    6355       2      83   53.2%    0.5%       136
 7 -      BATTERY      10       0      45    0.5%    0.5%        21
 8 -       BEEPER       6       0      83    0.5%    0.5%        17
```
```
gyro_sync = 1
gyro_sample_hz = 2000
gyro_lpf = OFF
pid_process_denom = 2
```

If I increase `set pid_process_denom = 1` (PID task running also at 2000 Hz), CPU load drops to 30%. **Very stange.**

```
# tasks
Task list          max/us  avg/us rate/hz maxload avgload     total/ms
 0 -       SYSTEM       3       0       9    0.5%    0.5%         1
 1 -         GYRO     249     202    1883   47.3%   38.5%     36052
 2 -          PID      78      36    1890   15.2%    7.3%      9602
 3 -        ACCEL     227     187     803   18.7%   15.5%     17217
 4 -     ATTITUDE      67      36      99    1.1%    0.8%       503
 5 -           RX      62      30      48    0.7%    0.6%       239
 6 -       SERIAL    1113       2      99   11.5%    0.5%      1295
 7 -      BATTERY       8       0      48    0.5%    0.5%        13
 8 -       BEEPER       6       0      98    0.5%    0.5%        16
```

```
gyro_sync = 1
gyro_sample_hz = 2000
gyro_lpf = OFF
pid_process_denom = 1
```

Does this reproduce on other targets?

hydra, 2016-11-05:
@lkaino thanks for your report - @digitalentity care to comment?

dkozionov, 2016-11-08:
The same problem on naze32 based board with all default settings. 500% CPU usage. The copter wont arm, and very strange GYRO and PID readings:

```
# tasks
Task list          max/us  avg/us rate/hz maxload avgload     total/ms
 0 -       SYSTEM       5       0       4    0.5%    0.5%         0
 1 -         GYRO       0       0   2147483647    0.5%    0.5%         0
 2 -          PID       0       0   2147483647    0.5%    0.5%         0
 3 -        ACCEL     127      96     498    6.8%    5.2%      4889
 4 -     ATTITUDE     370     339      49    2.3%    2.1%      1453
 5 -           RX      96      65      24    0.7%    0.6%       165
 6 -       SERIAL     276       2      49    1.8%    0.5%        59
 7 -      BATTERY      20       0      24    0.5%    0.5%        29
 8 -       BEEPER       5       0      49    0.5%    0.5%         6
10 -      COMPASS     130      99       4    0.5%    0.5%        51
11 -         BARO     133     102      49    1.1%    0.9%       521
13 -     ALTITUDE     262     141      19    0.9%    0.7%       253
```

DmytroPT, 2016-11-08:
It's happens on MultiwiiRewrite and 2,3. On LuxFloat is ok. Anybody can explain why?

Headcrawler, 2016-11-11:
same problem here on CC3D
on multiwii rewrite load of 500%+
With Luxfloat lower but still around 200-300% cpu load

Beneath taskdump on mwrewrite
# tasks
Task list          max/us  avg/us rate/hz maxload avgload     total/ms
 0 -       SYSTEM       4       0       4    0.5%    0.5%         1
 1 -         GYRO     252     161     883   22.7%   14.7%     38249
 2 -          PID     934     903     896   84.1%   81.4%    125305
 3 -        ACCEL     159     109     270    4.7%    3.4%      7528
 4 -     ATTITUDE     332     290      33    1.5%    1.4%      2048
 5 -           RX      99      67      24    0.7%    0.6%       434
 6 -       SERIAL   69294       2      14   97.5%    0.5%       271
 8 -       BEEPER       6       0      33    0.5%    0.5%        12

#

SkyCowboy, 2016-11-12:
Im having the same problem on all my naze32s after updating to latest fw they only work on luxfloat the others cause 600% cpu usage and cant arm

martinbudden, 2016-11-12:
@hydra , it looks like `pidInitFilters` is being called every time in the pid loop for MWREWRITE and MW23.

SandroGrzicic, 2016-11-12:
Looks like `pidInitFilters(pidProfile);` was added near the beginning of `pidMultiWii23` and `pidMultiWiiRewrite` by Hydra on 2016-09-19:

https://github.com/cleanflight/cleanflight/commit/2a7a768c1510195772e23e8951c625f362e04c71#diff-c43f5bbcc38711000682c7da30eef4d8R149

https://github.com/cleanflight/cleanflight/commit/2a7a768c1510195772e23e8951c625f362e04c71#diff-c43f5bbcc38711000682c7da30eef4d8R149

It was also added to Luxfloat:

https://github.com/cleanflight/cleanflight/commit/2a7a768c1510195772e23e8951c625f362e04c71#diff-b80362e82c3bcf8bdeb7cb976262d811R91

SkyCowboy, 2016-11-12:
Is the only way to fix this for now to downgrade to firmware version 1.13.0? Will this error get fixed and a new version made available anytime soon? Sorry for the presumably dumb questions but this is all way over my head I just joined this forum to find a solution to this issue. Thanks

SandroGrzicic, 2016-11-12:
My guess is that a fix will be made soon (I can do a PR but I have no idea if this is the correct fix to do) and presumably be available in a new patch version, 1.14.2, but that's just a guess, let's wait for someone "official" to reply.

Poitivin, 2016-11-13:
Just sharing I updated today from 1.13 (working fine) to 1.14.1 (naze32 rev5 with softserial+gps+telemetry) and in multiwii pid controller cannot arm. After reading this issue here and by just switching to pid controller LuxFloat, motors arm ok and everything seems ok.
Will keep an eye to see when 1.14.2 comes out to fix this

SkyCowboy, 2016-11-13:
yes hopefully someone important will notice this topic soon and fix asap as its clearly not an isolated issue.

SkyCowboy, 2016-11-13:
Also although it will arm and fly on Luxfloat (1.14.2) the cpu load on all my naze32s is around 30% and 60% on my CC3Ds when its just sitting there, whereas now Ive reverted them all back to 1.13 the CPU load is 0% on Multiwii rewrite or Luxfloat.

martinbudden, 2016-11-13:
@hydra , this needs fixing and a 1.14.1 release ASAP. The fix is straightforward, as I said in my previous comment.

Poitivin, 2016-11-13:
The issue happens also in 1.14.1 so I guess you wanted to say the fix release would be 1.14.2

martinbudden, 2016-11-13:
Thanks, corrected.

Poitivin, 2016-11-20:
Hi, any work being done regarding possible 1.14.2 to fix this issue on Naze?
Thanks

iss12, 2016-11-24:
same problem! :(

SkyCowboy, 2016-11-24:
this is a MAJOR BUG effecting everyone so I dont understand why there has been no 'official response' and an ETA given for a fix??? Maybe the title of the topic should be changed to "1.14 ONLY WORKS ON LUXFLOAT FOR NAZE32" and even though it works (just barely) on luxfloat it fried 2 of my littlebees the other day because I used the same PIDS as on 1.13!! I only raised the 'D' 4 points above default settings and 2 ESCs that were working perfectly for 4 months on 1.13 got smoked. So for now I would definitely suggest sticking to 1.13 on NAZE32!!

adrianbear, 2016-12-11:
Hi All,
I just had the same problem. Solved it by flashing with FULL ERASE (as recommended in the doco).
Version 1.14.1 now arms and "appears" to work correctly.

Hope this helps and may explain why no-one "official" is responding.

Adrian

SkyCowboy, 2016-12-11:
Ive tried at least twice on 3 different naze32 boards and 2 CC3Ds always with FULL CHIP ERASE and always same problem.. sticking to 1.13 since that works flawlessly

lkaino, 2016-12-13:
@SandroGrzicic explained the reason in his comment. The fix should be trivial. I removed the filter init call in my branch which solved the CPU load issue. Don't have possibilities to test the filter functionality atm.

TheSmartGerman, 2016-12-19:
Good Morning, I also currently have this problem too. 
I use the 1.14.1 (Release 11.11.) on CC3D.
Only LUXFLOAT gives me a CPU load ~60%-80%. 
If I enable "ANGLE" or any other flight mode it will jump too ~ 120% and i can't arm anymore.
What's your experience?
So far I'll downgrade to 1.13...

SkyCowboy, 2016-12-19:
Yes right now the only solution is downgrade to 1.13 I really dont understand why this hasnt been fixed or even acknowledged.. I can confirm today it doesn't effect my new SP racing F3 (clone) but Naze32 and CC3D dont work with 1.14.1

unitware, 2016-12-19:
Note that @andre92 noticed this problem even in 1.14.2-RC1

TheSmartGerman, 2016-12-22:
I testet 1.14.1 with my SP Racing Evo, CPU Load is ~20-30%.
For 1.13. on my CC3D the CPU Load is shown with 0% but cycle time is ~1000.
@SkyCowboy what do you mean with frying the ESCs on CC3D?

SkyCowboy, 2016-12-22:
Actually I can only say with certainty 1.14.1 can fry littlebee ESCs with Naze32 rev6 as I stopped using 1.14.1 on CC3D before it could do any more damage. I like 1.14.1 on my SP Racing F3 especially the new LED functions, also if using 1.13 you have to use the old cleanflight GUI to adjust battery voltage settings which is a pain.

hydra, 2016-12-27:
fixed by 93238bbee5b55a9792d116d0ada84bf819154d18

please test v1.14.2 RC3 when available.

thanks for bug reports and feedback everyone.

closed 2016-12-27 by hydra

max246, 2017-02-11:
Still happening with the latest firmware. I have a 500 or more % in the CPU. I am using Naze rev6 full

SkyCowboy, 2017-02-11:
1.14.2 fixed the CPU load prob on naze for me but still overheats my 20a littlebees like crazy so sticking to 1.13 (on naze) but running 1.14.2 on all my SP Racing F3s flawlessly

GilDev, 2017-02-14:
Same problem here on 1.14.2 with a Naze32.

`# tasks
Task list          max/us  avg/us rate/hz maxload avgload     total/ms
 0 -       SYSTEM       2       0       4    0.5%    0.5%         0
 1 -         GYRO       0       0   2147483647    0.5%    0.5%         0
 2 -          PID       0       0   2147483647    0.5%    0.5%         0
 3 -        ACCEL     170     139     499    8.9%    7.4%       377
 4 -     ATTITUDE     325     293      49    2.0%    1.9%        72
 5 -           RX      80      47      24    0.6%    0.6%         8
 6 -       SERIAL     330       2      49    2.1%    0.5%         4
 7 -      BATTERY      16       0      24    0.5%    0.5%         1
 8 -  HW_WATCHDOG       2       0       0    0.5%    0.5%         0
 9 -       BEEPER       6       0      49    0.5%    0.5%         0
10 -          GPS     372      59      49    2.3%    0.7%        13
11 -      COMPASS     127      59       4    0.5%    0.5%         3
16 -     LEDSTRIP     207     176      49    1.5%    1.3%        46`

6.00 load. Also serial's maxload jumps to 7000% sometimes but I guess this is normal.

hydra, 2017-02-14:
this is already fixed in master, we a bugfix release for this.

GilDev, 2017-02-14:
So why doesn't it works now? Should I open a new issue?
The problem is that tasks have low CPU usage but CPU is constantly overloaded.

atlaware, 2017-02-17:
With cc3d and cleanflight 1.14.1 the problem persist, with luxfloat pid system load at ~2.50 and NO arming
With Multiwii (rewrite and original) all ok, system load at 0.3 and correct arming.

Some news? I have just upgrade :(

Thank you!

stretford-chris, 2017-02-24:
This is definitely still a problem with 1.14.2 on a cc3d, cpu around 200+% and won't arm.  Switching to Multiwii rewrite drops cpu to around 30%.
I have gone back to 1.13.0 & this shows cpu at 0%, is 0% normal, I have never used cleanflight (or anything else) before?
When Hydra says "this is already fixed in master, we a bugfix release for this.", does this mean the fix has gone into 1.14.2 or is this for a later version?

stephenvibes, 2017-02-25:
Glad I found this thread. Thought (as a newbie) I was going mad.

I'm having the same problem with Cleanflight 1.14.2, on an Eachine Blaze 185.
Odd numbers for the GYRO/PID rates.

Hoping for a working update soon.

'# tasks
Task list          max/us  avg/us rate/hz maxload avgload     total/ms
 0 -       SYSTEM       5       0       4    0.5%    0.5%         4
 1 -         GYRO       0       0   2147483647    0.5%    0.5%         0
 2 -          PID       0       0   2147483647    0.5%    0.5%         0
 3 -        ACCEL     173     140     498    9.1%    7.4%     41765
 4 -     ATTITUDE     330     298      49    2.1%    1.9%      8149
 5 -           RX      92      54      24    0.7%    0.6%       964
 6 -       SERIAL   83323       2      49  408.7%    0.5%       727
 7 -      BATTERY   40077      14      24   96.6%    0.5%       228
 8 -  HW_WATCHDOG       4       0       0    0.5%    0.5%         0
 9 -       BEEPER       8       0      49    0.5%    0.5%        41
11 -      COMPASS     130      99       4    0.5%    0.5%       324
12 -         BARO     136     105      49    1.1%    1.0%      3282
14 -     ALTITUDE     257     140      19    0.9%    0.7%      1633

'# status
System Uptime: 518 seconds, Voltage: 115 * 0.1V (3S battery - OK), System load: 5.52
CPU Clock=72MHz, GYRO=MPU6500, ACC=MPU6500, BARO=MS5611, MAG=HMC5883
PIDd: 0, GYROd: 0, I2C Errors: 0, registry size: 53<br />

GilDev, 2017-02-26:
@stephenvibes

Yeah my results were also with an Eachine Blade 185. I only tried with Luxfloat. Going back to 1.13.0, it works well!

max246, 2017-02-26:
Found out that my board had bad gyro or another sensor, it was throwing i2c errors sometimes. Since I swap to a new one, all issues are gone

hydra, 2017-03-01:
thanks for the update.