cleanflight/cleanflight #2480: [OSD] On-screen amperage meter readings don't seem to be correct

SandroGrzicic, 2016-11-06:
On the SP Racing F3 OSD/PDB, after wiring up everything and testing that the OSD works, everything seems correct, except the on-screen current meter which doesn't seem to work properly. With the main voltage supply connected, it constantly shows 0.07A, sometimes going up to 0.10A. With only the USB supply connected, it shows 0.05A. These values cannot be correct since the PDB/OSD board itself, VTX, camera, FC (SP Racing F3 Evo) and an RX (FrSky XSR) surely consume more than 100mA. After arming the quad and starting motors, the current readings still don't change.

I have tested this with two separate boards and the same behaviour persists. Both are freshly flashed with the latest Cleanflight v1.14.0. Please close this if this is a hardware issue, I am opening it in case it's a software issue.

element51, 2016-11-08:
i have the exact amperage issue with my SPRacingF3OSD, voltage is reading ok.

SandroGrzicic, 2016-11-11:
Any updates on this? I would like to try fixing the issue, but without knowing where approximately in the codebase to look at, it's a bit difficult (I have already checked the amperage sensor code, I don't see anything obviously wrong there).

Some random ideas (assuming the issue is in Cleanflight and not in the board itself):

- maybe the "raw" ADC readings come from the virtual current sensor? But that doesn't explain the random jitter, nor the difference between USB plugged/unplugged.
- maybe the current sensor's current readings are corrupted by Betaflight FC data? But that doesn't explain seeing the exact same behaviour when the FC is disconnected (surely the PDB board doesn't need the FC to display the amperage).
- `adcGetChannel` might be using the wrong channel (`ADC_AMPERAGE`)? I noticed the same method is used to get the voltage readings; maybe instead of the current, the code is actually reading one of the voltage ADCs (e.g. the 5V or 12V voltages, the main battery voltage, or the 3.3V reg voltage)?
- the wrong ADC driver is being used?

hydra, 2016-11-17:
@SandroGrzicic @element51 sounds like a wiring issue or manufacturing issue.  Contact your retailer directly to get this resolved.

closed 2016-11-17 by hydra