cleanflight/cleanflight #2482: HMC5883 external compass not working with 1.14.0 or 1.14 RC-2

Nihilistic, 2016-11-06:
After updating to 1.14 from 1.13 (checked full chip erase) on my Naze32 Rev5, the external HMC5883 compass contained in my UBLOX NEO-M6N GPS puck doesn't work (mag icon isn't lit in the CF configurator 1.2.4 and the mag calibrate button is greyed out). I'm also using an external BMP280 barometer soldered to the same i2c pads. Flashed back to 1.13, and it all works fine.

hydra, 2016-11-07:
are you seeing any I2C errors?

@digitalentity probably related to the I2C changes?  Can't see any other reason for this.

Nihilistic, 2016-11-07:
No I2C errors. The external BMP280 worked fine on the bench. I tried disabling the BMP280 via the CLI command line to see if the external mag would then work. It didn't. Either would work if the other wasn't enabled in iNAV 1.21.

kevin678, 2017-01-10:
my HMC5883L is not wirkung too, but my HMC5983 work very well with this Software.

I used the CC3D Board with the newest firmware

pkedvessy, 2017-01-10:
I have the same issue with Skyline32 Advanced onboard HMC5883 sensor. It is working as expected with 1.13 but 1.14.x can not recognize the mag sensor. No I2C errors...

willanth, 2017-01-15:
I am using a NAZE32 Rev 6 board, and also have issues with the mag sensor.  It (the mag sensor) starts intermittently when I power it up, either through USB or with a pack.  The only reliable way to make the mag sensor start and be detected is to either flash firmware to the unit or "save and restart" under the configurator software.  I have blackboxed several flights, and out of 15 flights only 1 had mag data logged.  I have tested this on 1.14+ firmware revisions, but not previous versions yet.  My Baro sensor (BMP280) is working reliably and always starts and logs.  I haven't had an opportunity to sniff the bus and see what's going on, but from previous hardware experience some sensors are more tolerant of issues than others and I wonder if the HMC is not liking something that's changed in the codebase?  Happy to contribute how I can to get this issue fixed.

closed 2017-02-25 by hydra