cleanflight/cleanflight #2484: 1.14 + Naze32 rev5 Freezes after arming with Telemetry enabled

DopeyOne, 2016-11-07:
When using 1.14 on my Naze32 rev5 with OS125 + PPM + Telemetry (UART1 telemetry set to FrSky @ 9600baud)

If I arm the controller, I can no longer disarm and/or control the motors (looks like the FC locks up)

If I remove the UART1 telemetry configuration, board functions normally when armed

Replicated the same setup on 1.13 and I do not experience the issue.

hydra, 2016-11-07:
please supply config dump and we'll investigate ASAP.

DopeyOne, 2016-11-07:
[114_Naze32_Broken.txt](https://github.com/cleanflight/cleanflight/files/575204/114_Naze32_Broken.txt)
[114_Naze32_Working.txt](https://github.com/cleanflight/cleanflight/files/575205/114_Naze32_Working.txt)

DopeyOne, 2016-11-07:
Very basic/minimal configs...

But if UART1 is set to FrSky @ 9600, then arming the board causes it to freeze.
Disable FrSky Telemetry on UART1 and it doesnt freeze.

FW update was done with full chip erase, so there shouldn't be any legacy configs causing issues.

DopeyOne, 2016-11-07:
[113_Naze32_Working.txt](https://github.com/cleanflight/cleanflight/files/575219/113_Naze32_Working.txt)

Working 1.13 config with Telemetry on UART1

Poitivin, 2016-11-13:
I have the same setup you have and updated today my firmware from 1.13 to 1.14.1 and cannot arm in pid controller multiwii rewrite. LuxFlot arms ok and I have the same exact config between these two pid controllers with the exception of pid values and rc values...

martinbudden, 2016-11-13:
@Poitivin , your issue is a duplicate of https://github.com/cleanflight/cleanflight/issues/2479

Poitivin, 2016-11-13:
Thanks. It seems to be yes. I´ll keep an eye on #2479

Ryanpolo, 2016-11-22:
just wanted to add. i too had this problem. naze32 rev6. using 1.14 at idle was 600% cpu. reflashed to 1.13

lackr007, 2016-11-24:
I can also confirm that this FC lock up issue on 1.14 on 2 different naze32 rev5 + PPM + telemetry + LED. 1.13 works fine.

Ahimoth, 2016-12-12:
Same here. 1.13 works fine with FrSky telemetry enabled on naze32 rev5 and rev6. Updated to 1.14 with full erase and now FC appears to lockup immediately after arming when FrSky Telemetry is enabled. Disable the telemetry and it works fine.

hydra, 2016-12-27:
deleted unrelated comments.

hydra, 2016-12-27:
bug confirmed, issue is with the DMA handler initialisation when the SHARED UART port is re-opened for use as a telemetry port after being already used by MSP.

hydra, 2016-12-27:
fixed with this workaround.

6b7ac227b5c073560f5f0df534e46385ae099f84

hydra, 2016-12-27:
please test 1.14.2 RC3 when available.

closed 2016-12-27 by hydra