cleanflight/cleanflight #2485: CJMCU target, gyro with no reading in 1.14.0

myly10, 2016-11-07:
Still works fine in 1.13, but once flash to 1.14 it stops working.

Gamesharkster, 2016-11-15:
I'm having the same problem. When I fall back to 1.13, I lose vBatt monitoring. Any ideas on how to fix this issue?

hydra, 2016-11-17:
debug it? 😄 

but seriously, thanks for the report. hopefully someone can look into it.

bbfishyang, 2016-11-29:
Probably the MPU6050 INT pin didn't work. In 1.14, MCU won't get gyro data until receiving INT signal from MPU6050.
You could try to add "#define DEFAULT_GYRO_SYNC 0" in target.h. This will force MCU to get gyro data periodically.

Gamesharkster, 2016-11-29:
Too late for me guys:p I just switched to Betaflight. It has worked perfectly so far.

hydra, 2016-12-28:
fixed in f06975c87b0271827f779ad74e8ec746272c388f

please test v1.14.2-RC3 when available.

closed 2016-12-28 by hydra