cleanflight/cleanflight #2488: Restore to 1.2.3

romanator, 2016-11-08:
The backup feature was removed for 1.2.4.  How do I downgrade to v1.2.3 on my Mac?  I need the backup feature to restore my drone settings.

andre92, 2016-11-09:
I assume you mean the configurator. Download the version you want as zip: https://github.com/cleanflight/cleanflight-configurator/releases
extract it, and install it manually into google chrome (load unpacked extension)

downtown54, 2016-11-13:
Help!! How do you install manually into Chrome?

andre92, 2016-11-14:
@downtown54 Download the Configurator from https://github.com/cleanflight/cleanflight-configurator/releases (mind the version you want) and extract the archive in a new folder.
Next, visit chrome://extentions and enable developer mode. Then there will be a button to load an unpacked extention. Then find your unpacked folder of Cleanflight Configurator. It should install itself from there.

downtown54, 2016-11-14:
Thanks Andre

closed 2016-11-17 by hydra