cleanflight/cleanflight #2490: strange Blackbox data v1.14

xuyuan, 2016-11-11:
My copter flies fine, but blackbox gives strange data (i.e. battery value less than 1v, the heading is completely wrong)
Everything was fine in 1.13.0.

PS: Tested with NAZE

hydra, 2016-11-17:
need more information on that, e.g. steps to repeat, config dumps, logs, etc.

xuyuan, 2016-11-18:
For example, in this log [LOG00025.TXT](https://github.com/cleanflight/cleanflight/files/600242/LOG00025.TXT)

my quad was flying ok, it was hovering most of the time, but from the log file
* the battery voltage is about 1v
* the orientation is totally wrong in the picture below:
[image]

DengueTim, 2016-12-05:
I haven't been able to get blackbox data over serial with versions 1.14 and 1.14.1.   Worked as expected in 1.13.

xuyuan, 2016-12-06:
@DengueTim that's another bug, I have fixed and made pull request 26 days ago:
https://github.com/cleanflight/cleanflight/pull/2489

I wonder how many people use cleanflight 1.14 with blackbox, since the configuration breaks...

closed 2017-02-25 by hydra