cleanflight/cleanflight #2492: motor won't rev up after throttle cut

Pedroalbuquerque, 2016-11-12:
While flying in horizon mode, is I throttle up (jump in air), cut the throttle and while descending, try to throttle up again, frequently one or more motor do not rev up leading to an uncontrolled crash.

My setup:
Cleanflight 1.13
Board : Flip32+ (with baro + mag) from bangood
Radio FrSky via PPM
Bluetooth on UART1 for config
OSD on UART2 for FPV
 Esc ZTW 12A Opto with BLHeli 16014501 ( dumpen light active , all defaults)
Motor RCX 1804/2400Kv
Prop 5x4 or 5x3 (not affecting behaviour)
Min throttle = 1050
Max Throtle = 2000
Min command = 940
Oneshot125 = ON
Motor stop = ON
Disarme motor = ON
Arm switch = AUX2
I don't have a Blackbox divide... :(

**1st Assumption: Motor or esc cannot revup from stop position**
Test:on the bench tried to simulate by bringing throttle to low position after having motor spinning, and  bringing it up again
Result: none of the motores fails after numerous attempts. Only fails while flying, not on the bench.

**2nd Assumption: Brownout** 
Test: Tried several baterias:
Result: only fail while flying never on hand or the bench , does not change with battery

**3rd Assumption: Loop rate too high, processor does not have time to process Radio input while trying to level**
Test: raised loop time from 2000 to 2500 to 2800
Results: problem persists

**4rd assumption: air flowing from back of prop does not allow the motor to recoup while rotating backwards**
Test: setup a air drier under one motor full power(no heat), Started with throttle up, cut the throttle till motor stop and revue again
Result: no failures, even if motor started to rotate backward due to air blow, motor could still revue and rotate on correct direction. NO failures.

**5th assumption: there is some protection the software agains revue after motor stop?? **
test: set motor stop = OFF
Result: no more failures !!, same result if AIR mode is on, diference being that min rev with air mode is higher then simply having motor stop OFF.

WHY ???
Is there any protection on software? is it related to Baro somehow ?
Is there a way I can have motor stop after arm, but not getting "stucked" when going throttle to min in mid air ?

Still could not simulate problem on the bench or with quad on my hand even trying shaking it while doing the throttle test.

ledvinap, 2016-11-12:
My first guess would be **4..

Pedroalbuquerque, 2016-11-12:
Test result does not confirm, do you propose some other test ?

andre92, 2016-11-12:
Try raising minthrottle to 1070. Props encounter more resistance in air than on the bench, especially when descending. What works on the bench, does not nessecarily work in the air.

Pedroalbuquerque, 2016-11-12:
Thank you André, that makes sense and aligned with my positive result, i will try that.

hydra, 2016-11-17:
please use rcgroups.com for support, this is an issue tracker for bugs/ideas/features, etc.  review contributing guidelines before creating issues please.

closed 2016-11-17 by hydra