cleanflight/cleanflight #2494: Accelerometer filtering and vibrations, angle/horizon not working: possible fix

mamalala, 2016-11-13:
In chat it was suggested i make a PR for this, forgot that i already have a GH account, and now decided to open an issue instead.

When there are vibrations picked up by the ACC, in 1.14.1 (and 1.14.0 as well, but this is referenced to the 1.14.1 tree) a level-sitting copter will show increasing tilt in the configurator when the motors are running. This will result in the copter getting silly in the angle and horizon mode. The ACC readings also show heavy noise, +/- 1g or even more.

Since 1.13 did not have that issue, and showed less noise, i decided to dig into the sources and see what changed, to try to come up with a quick fix. Here is what i found out, and how i fixed it, although it very likely does not resolve the root cause in the cause, i think.

First, it seems that pt1FilterApply4() in common/filter.c is  not working correctly. To make that issue visible, first change .desiredPeriod in fc/fc_tasks.c for the ATTITUDE task to something higher, like 250 Hz:

```
    [TASK_ATTITUDE] = {
        .taskName = "ATTITUDE",
        .taskFunc = taskUpdateAttitude,
        .desiredPeriod = TASK_PERIOD_HZ(250),
        .staticPriority = TASK_PRIORITY_MEDIUM,
    },
```

With only that change, nothing will really happen regarding the issue. Now, in common/filter.c at line 52 remove the usage of filter->dT and have it use the dT given as the functions argument instead (like it was in 1.13):

`filter->state = filter->state + dT / (filter->RC + dT) * (input - filter->state);`

With that, the issue is basically gone. It seems that somewhere the filter->dT is set wrongly, and subsequent calls to the function use that different dT, making the filter not working  correctly. I did not dig deeper into that one, so i have no idea where/why that happens.

Also, when TASK_ATTITUDE is kept at 100Hz, even with the above "fix" to filer.c, the problem remains (using all default values in the FC config). So  it had to be made run faster to actually resolve the problem, causing a noticably higher CPU load.

Because of that i checked a bit further about the changes between 1.13 and 1.14.1. In 1.13, it seems that the task for the ACC was just one, whereas in 1.14 there is TASK_ACCEL and TASK_ATTITUDE. So i made two other changes, one is to fc/fc_tasks.c, commenting out the TASK_ATTITUDE section while changing TASK_ACCEL to:

```
    [TASK_ACCEL] = {
        .taskName = "ACCEL",
        .taskFunc = taskUpdateAccelerometer,
        .desiredPeriod = TASK_PERIOD_HZ(500),
        .staticPriority = TASK_PRIORITY_MEDIUM,
    },
```

And also i chnaged fc/cleanflight_fc.c so  that taskUpdateAccelerometer() now handles both, while commenting out the call to imuUpdateAttitude() in taskUpdateAttitude():

void taskUpdateAccelerometer(void)
{
    imuUpdateAccelerometer(&accelerometerConfig()->accelerometerTrims);
    imuUpdateAttitude();
}

void taskUpdateAttitude(void) {
//    imuUpdateAttitude();
}

So far, all this fixed the described problem for me, while keeping CPU load down. Obviously i am new to the CF codebase, and i did not dig into everything, so i'm pretty sure that these fixes are mainly handling the symptoms, but not the root cause in the code. Especially the issue in filter.c seems to be a real bug.

With  the severly limited testing i could do on the quad, there seem to be no adverse side effects due to these changes, they simply fixed the issue for me.

Hopefully this will point the dev's into the right direction to look into.

Greetings,

Chris

andre92, 2016-11-14:
see this PR: https://github.com/cleanflight/cleanflight/pull/2477
the real issue is the acc filter being run at slow rate. Filtering should always be done at same speed as sampling. The current way, 9 out of 10 samples are thrown away since acc filter is run in task attitude. Betaflight fixed this some time ago by doing the acc filtering in task_accel.

mamalala, 2016-11-14:
I'm not convinced that just increasing the call frequency is the solution. At least for me that did not help at all, which is why i found the filter->dT vs. just dT (the function argument) issue.

I was wondering as well about the two different rates for ACC reading and attitude correction stuff. In case the raw ACC values are not used anywhrere else (at least in imu.c only accSmooth is used), it is rather silly to have it that way. But then, i did not look at other parts of the code to check that.

andre92, 2016-11-15:
The difference between Betaflight and Cleanflight is the way sensor data is treated.
Betaflight runs the filter after every sample collected, while Cleanflight only runs the filter when a sample is needed. As a result, not every sample goes through the filter. The IMU may be only interested in the filter output, but 100 samples per second into the filter won't do, it never did. See Cleanflight 1.12 and related issues.
Since the filter is run only at 100hz (in the IMU) only one out of 10 samples makes it into the filter, the rest is lost. Betaflight has fixed this by running every sample through the filter. They do not call the filter in the IMU but in the task that samples the ACC, so the filter output will be cleaner.

putersdcat, 2016-11-16:
I also have experienced a dramatic fluctuations of +/- 1.0g or more in ACCEL without propellers attached just calibrating motors at my desk. Initially I attributed the problem to a faulty MPU-6500 sensor in my Naze32 Rev6 clone, so today I switched over to a SP Racing F3 Delux clone with an MPU-6050, thinking that would solve the issues.

Just as a control I waited on upgrading the SRF3 until I had done some bench testing. So I started with Cleanflight 1.12.0, spinning up the motors with no propellers attached, the max ACCEL I could hit was +0.14 xyz. So I felt that my suspicion about the faulty MPU6500 in the Naze32 Rev6 was confirmed. But then after I upgraded the SRF3 to 1.14.1 i could immediately see from some quick testing (spinning up the motors on my desk) that it was un-flyable! With ACCEL readings of +\- 2.0 g. I know from the Naze32 that trying to fly like this will end in immediate flipping nose dive. 

So flashed back to 1.12.0 and reset to factory defaults, then restored my basic receiver settings, modes, etc. and again in testing I was back to very reasonable and flyable +\- 0.14g max! 

I have since tried 1.14.0 and it behaves identically to 1.14.1 as far as the terribly inaccurate ACCEL readings. I can also confirm that a quick flash over to beta flight 3.0.1 did not have the ACCEL issues. So for now I am sticking to CF 1.12.0, but I would like to point out that this bug report above is something significant that persists across multiple FC hardware and at least the two different MPU HW types.

Also of it helps here are some dumps of my configs for both boards, 1.12.0 working and 1.14.0 / 1.14.1 not working on same hardware.

https://www.dropbox.com/sh/3fzcn6qaae798qi/AABuJ0zjqCInI-k7RrzUc6QTa?dl=0

andre92, 2016-11-17:
Interesting. I found Cleanflight 1.12.1 only samples the acc at 100Hz but it runs the lowpass filter at 1kHz. So that appears to work better than Cleanflight 1.14, which samples at 1kHz and filters at 100hz.
Betaflight does both the sampling and filtering at 1kHz, which seems to work best. The difference is that Betaflight moved the filtering out of the IMU and does it at the same rate as sampling.

hydra, 2016-11-17:
thanks for the feedback.  hopefully someone will look into this and create a PR soon so we can release v1.14.2 with a fix for it.

andre92, 2016-11-18:
I made pr #2477. It executes the acc filter after each sample, instead of in the IMU, similar to Betaflight. Only difference is that it is still PT1 and not biquad, but it should work.

mamalala, 2016-11-19:
I sat down and modified the code so that it behaves like Betaflight w.r.t.  accelerometer handling/filtering. Just created a PR, #2505, for that. Hope it helps.

I have not touched anything in the unit-test stuff, dunno if that may cause issues or not.

hydra, 2016-11-29:
fix merged to master and to v1.14.2

closed 2016-11-29 by hydra