cleanflight/cleanflight #2499: SPF3 Board Won't Arm?

Jahigs, 2016-11-15:
Hi! practically a newborn when it comes to quadcopters, but i managed to get my GB-190 kit set up and it responds to cleanflight. I've linked my receiver to cleanflight and i have my sticks set correctly(Running an FS-R6B on PWM) . My problem lies with arming the board I believe?  I cant get any response when moving my sticks and i've been told arming an SPF3 is just holding throttle down to the right and you should see a green light? anyways, if anyone knows how to fix this arming issue that'd be great, i've been at it for several hours now with no luck. Also I have no way to set up aux channels so i cant just move the arming to an aux switch :(

Gamesharkster, 2016-11-15:
I'm having the same issue with my BeeRotor F3. If you fall back to 1.30 firmware you'll be able to arm, but you'll lose battery monitoring.

Jahigs, 2016-11-15:
I fell back to 1.30 and still no luck with arming :\ thanks for trying though

Jahigs, 2016-11-15:
UPDATE: Added an arm switch to aux 1 and it did the trick

closed 2016-11-15 by Jahigs