cleanflight/cleanflight #2506: Betaflight with Revolt F4

Maxthepersonz, 2016-11-19:
Does anyone know if this is possible? Just curious to throw the Raceflight board on Betaflight and see what I think. 

What firmware do I flash? Or none available yet?

-Max

martinbudden, 2016-11-19:
Betaflight 3.1 will ship with a Raceflight Revolt target. See:

https://github.com/betaflight/betaflight/blob/master/src/main/target/REVO/REVOLT.mk

You can build it yourself now, but remember it is pre-release with all the risks that entails.

BTW, this is the Cleanflight not the Betaflight repository.

closed 2016-11-22 by hydra

William3170, 2017-02-11:
I just finished flashing Betaflight 3.1.5 to my Revolt FC. Works great, you just need to use Zadig to change driver. Install revolt on CDC driver and you should be able to connect FC after flashing BF Firmware.