cleanflight/cleanflight #2511: SP racing F3 mini dead

jollyrogeruk, 2016-11-25:
Hi
My New FC is dead.
I have it installed and it was all working as it should (or appeared to be). Latest Firmware flashed, esc's programmed etc. It setup ok, motors tested and calibrated ok. Radio setup ok and started with default pids. All settings saved and bench run ok.
Battery disconnected and reconnected ready for test flight and it's dead no blue or red LED only a solid green LED (5V). Plus it now won't connect with cleanflight. I belive the red LED is the 3.3v ok, so I guess the 3.3v is missing for some reason.
I have checked all joints etc, no shorts, all wiring looks ok. I have disconnected the FC and tried powering with USB only and still only one green led. Any ideas ?

Setup.....
Littlebee 20A esc
Emax RS 2205
Spektrum satellite rx

Any help would be very much apreciated

Sigius, 2016-11-27:
Go to Boot mode and try to reflash

xvoland, 2016-12-25:
@jollyrogeruk I protect FC like this https://youtu.be/lSCjASAPoEA
I use capacitor 200uF/50V and no more problems

closed 2017-02-25 by hydra