cleanflight/cleanflight #2514: [OSD] RSSI is always zero

SandroGrzicic, 2016-11-27:
Latest (master) CF flashed to the OSD shows a RSSI of zero no matter what the actual RSSI is sent from the FC.

Looking at `osdElementRender_rssi.c`, we have:

    uint16_t rssi = (uint16_t) dataFn();
    tfp_sprintf(elementAsciiBuffer, "%3d", (rssi / 1023) * 100);

But rssi is always between 0 and 1000 since in `msp_client_osd.c` we have:

    fcStatus.rssi = scaleRange(sbufReadU16(src), 0, 1023, 0, 1000);

So won't any rssi between 0 and 1000 after being divided by 1023 be zero?

Since `rssi` is 0 to 1000, isn't it enough to have:

    tfp_sprintf(elementAsciiBuffer, "%3d", rssi / 10);

?

hydra, 2016-11-29:
yes, this was due to unmerged changes for the spracingf3neo board support which handled this differently.  Likely we will need to address this again when that branch is merged in.

hydra, 2016-11-29:
See #2517

SandroGrzicic, 2016-11-29:
Since my PR has been merged I'll close this issue, a new one can be raised if there is any RSSI issue later on.

closed 2016-11-29 by SandroGrzicic

hydra, 2016-12-06:
see cb6c86149999475a98d471e1f4270dbc49ce0eb7