cleanflight/cleanflight #2520: EXTI mux

jflyper, 2016-11-30:
I'm really sorry to bring this up now.

Is multiplexed IRQ dispatcher gone?

ledvinap, 2016-11-30:
Can you be a bit more specific, please?

jflyper, 2016-11-30:
Apologies; I was bit hurrying when I wrote the issue.

IIRC, there was EXTI line to IRQ handler dispatcher with small array of { irqnumber, handler } so that multiple pins can be that are multiplexed onto single EXTI can be registered and called back (e.g., PA2, PB2, PC2, PD2, ...).

Is this mechanism/service gone now? Or it wasn't there after all.

All I see now is extiChannelRec[16]...

ledvinap, 2016-11-30:
Old EXTI code was replaced with my implementation, it's part of IO abstraction.
Currently there is single handler per EXTI pin (and non-pin EXTIs are neither supported nor used yet), I did not see reason to attach multiple drivers to single pin. EXTI is multiplexed, so only one pin can be connected to EXTI.
Do you have use case where multiple handlers are necessary? 

`EXTIConfig(IO_t io, extiCallbackRec_t *cb, int irqPriority, EXTITrigger_TypeDef trigger);` handles EXTI setup for given pin, all needed parameters are passed as arguments, everything other is inferred internally.

`extiConfig_t` is provided for partial backward compatibility with code that used old pin access routines.

jflyper, 2016-11-30:
So, to recap, current EXTI support is per (EXTI) pin and not per EXTI line?

ledvinap, 2016-11-30:
EXTI handler handler is attached to given pin. 
Due to design on EXTI peripheral, only 16 pin handlers are possible, so corresponding EXTI line is used and not available for pins on other ports.
Pin collision is not checked, so registering handler for PA1 and PB1 has undefined results (last handler wins, old ones are silently replaced)

jflyper, 2016-11-30:
I've got it. Thanks, as always!

closed 2016-11-30 by jflyper