cleanflight/cleanflight #2521: FC lock due to UBLOX M8N overload?

EricBNL, 2016-12-02:
I want to share my experiences with the community regarding using an UBLOX NEO-M8N GNSS receiver on a 19200 soft serial port of a Naze32 rev.6 Flight controller. CF was configured for auto configuration in UBLX mode with SBAS enabled.
I configured the above configuration and soon noticed lock-ups of the FC after running great for several minutes. A power cycle is needed to get in control again. Fortunately my quad had not left the workbench yet so no harm done :-)

After investigation I noticed several forms of overloading of the FC connection:

1. The M8N is a multiple GNSS receiver with up to 32 channels. It can simultaneously lock on to GPS, Glonass and Galileo satellites. When the receiver is connected with a FTDI serial adapter to a PC over a 19200 baud link (to simulate the soft serial connection) and the UBX>MON>TXBUF message is enabled a buffer overload can be observed. There is too much data to be send over the serial link at 19200. I managed to configure a stable, non overloading 4Hz output with the following settings:
Messages	(interface, message drop factor)
MSG > 01-04 NAV-DOP	UART1,1
MSG > 01-02 NAV-POSLLH	UART1,1
MSG > 01-06 NAV-SOL	UART1,1
MSG > 01-30 NAV-SVINFO	UART1,10
MSG > 01-21 NAV-TIMEUTC	UART1,1
MSG > 01-12 NAV-VELNED	UART1,1
MSG > 01-03 NAV-STATUS	UART1,20
	
Rate	
RATE > Time Source	1 - GPS time
RATE > Measurement Period	125 ms
RATE >  Navigation Rate	2 cyc

These settings are not pushed on the GNSS receiver when CF is in gps auto configure mode. It tries to enable a 10Hz output which is obviously too much.

2. When observing the output, the message "NAV-SVINFO" is by far the largest contributor. The message size is dependent on the number of active channels and activated GNSS systems. Of course, I want to use as much satellites as possible. When observing the message size for "NAV-SVINFO" it increases along with the number of satellites in use. The maximum message size observed by me is 584 bytes. Possibly even larger values are possible when more satellites are in view. The UBlox protocol description manual (u-blox8-M8_ReceiverDescrProtSpec_(UBX-13003221)_Public.pdf) page 292, chapter 30.18.18 details the constitution of this message. For every channel 20 bytes are added to the message.

 I read somewhere significant smaller message size is accepted. It's this overload that is likely to be the root cause for the lockups.

3. I live in Netherlands and most of the country is below sea-level. When observing altitude values in CF configurator I see values of 65000+ meters. I looks like a wrong interpretation of the altitude message of the GNSS receiver. The quad might think it is flying at 65km altitude and drop throttle to maintain the alt hold value. Would it be possible to use altitude above WGS84 model instead of the current calculated height above sea level?

I would like to know if anyone can confirm my findings and possibly point me to a solution that will maximise the possibilities of the GNSS receiver and does not lock the FC.

Happy flying!

andre92, 2016-12-03:
probeer de Inav vork van Cleanflight. Die hebben GPS veel beter voor elkaar.

unitware, 2016-12-05:
@andre92 would you like to translate that? I think I understand, but Swedish is only somewhat influenced by dutch, the details get lost..

EricBNL, 2016-12-06:
@unitware the requested translation: "Try the INAV fork of Cleanflight. They perform much better. I have never had any luck with GPS on a softserial port. I use with Naze a ppm receiver and telemetry on soft serial and gps on uart2."

@all 
Thank you all for your response. I have had a peek inside the source code of INAV. However different, it still does not support the UBLOX Neo M8N. It only supports the less complicated older versions.

In my opinion there is a need for more room to store the NAV-SVINFO info from multiple GNSS receivers like this NEO M8N. It is also my belief that softserial at 19200 baud is not the limiting factor for my issue. When I carefully select the needed messages at the appropriate rate there is no overloading of the serial connection. Since I am not capable enough to reprogram and recompile the source code I would like to ask the programmer for the following adjustments:
1. accommodate more room for the NAV-SVINFO message. Appr. 800 bytes will do.
2. use updated messages. Some messages are deprecated or replaced by UBLOX.
3. Adjust or correct the altitude reading for below sea-level terrain. A relative altitude above take-of point would also be fine since that's what we want to know when flying :-)
4. accomodate for the 32 channels this type of receiver offers.

Thank you all.

hydra, 2016-12-06:
packet size is large on the M8N receivers, likely the FC using softserial is overloaded.  We're already aware of the M8N SVINFO sizes, the code in iNAV which addresses this will be merged in due course.

closed 2016-12-06 by hydra

unitware, 2016-12-07:
I'm leaning towards that this issue should be left open as a starting point to the issue https://github.com/cleanflight/cleanflight/issues/1621 that handles the complete navigation and GPS code (which I am guessing @hydra is referring to) or is there another issue/task that I missed?

EricBNL, 2016-12-07:
Besides the memory and code merge issue in #1621, there still is the altitude calculation (interpretation) which is faulty with unknown consequences for altitude hold and return to home feature at or below sea-level terrain. Choosing other messages and/or rates could be part of an improvement strategy regarding GNSS support in CF and solving this issue as well. No CLI is fine with me, but some configuration freedom such as manual GNSS configuration and soft-serial port speeds should be preserved in some way.