cleanflight/cleanflight #2522: Flip on take off with cc3d

Addicted5, 2016-12-02:
Hi, i finished my 1st quad build.
Here is the [picture](http://imgur.com/a/7TEET).

But it's trying to flip when i try to take off.
Here is Blackbox [log](https://mega.nz/#!HBAymCIL!VSnwKPKzxHYQpiep59gDH5LDTijGM9qOW4koJOB09Dk) without propeller.
This is [log](https://mega.nz/#!6UBSgRLR!GRyP2o16cIeQRcnULAQqXhUUjTxgSs5OqD_Kvu36TJA) with propeller where throttle is constant & left 2 motor increased speed after like 22s.

CC3D is correctly mounted because when i move quad , it also moves same way in cleanflight gui.

andre92, 2016-12-03:
I would very much like to see your CLI dump, but here's a little checklist:
- board orientation correct in GUI
- accelerometer calibrated on a level surface
- compass calibrated (if you have one)
- motors are on the correct output (use motor tab to spin them up 1 by 1)
- motors spin in the direction indicated by the picture in the Motors tab
- props are on correct (your picture shows correct props)
- props are balanced (definitely balance props over 7 inch, yours look like 10)

According to your blackbox log, there is either a lot of vibrations or your P and D gains are too high.
(I suspect the first)
On a 450 with 10 inch props you should use gyro_lpf = 188hz, gyro soft filter 50-80hz, and Dterm filtering 30-50hz. With this kind of Dterm oscillations, you are going to burn a motor. For my quadcopter, 60 for gyro soft lowpass and 30 dterm lowpass works fine.

Addicted5, 2016-12-03:
my quad is like 900gram with 1400kv motor with 10*4.5 prop & 3S lipo.
(I came to know that i needed to buy 1000KV motor for 3S & 10" prop.)
But the seller selling 1400KV motor & 10" prop in combo anyway.
May be i need to limit max throttle to prevent burning of motor.( any way i haven't takeoff & ran motor max 40s, so haven't burned my motor, need your suggestion about this)

I reset all setting.So i'll again reconfigure all my set as previous & will post dump+log tomorrow.

Currently my max throttle is 1850.May be i need to lower max throttle more.Need suggestion on this.
I don't need acro performance. I just want to learn basic with this build.(That's why i don't want to buy 1000KV motor for this build as i planed to make another quad & 9" propeller seems hard to find. anyway will 8" prop with 3 wing able to lift 900gm?)

andre92, 2016-12-04:
Yes, those settings are in CLI. 
The settings are gyro_lpf, gyro_lowpass_hz and dterm_lowpass_hz (exact names differ per version)
What type of motors and ESCs do you have? It is not only the kv but also the size. A 1400kv 2212 isn't happy with 10 inch but a 1400kv 2216 may do just fine.
8 inch 3 blade will work also, as a matter of fact I switched to those and I like them better than 10 inch 2 blade. They produce less lift but I think they respond better in flight.
You should first set gyro_lowpass_hz to 80 and dterm_lowpass_hz to 50 and see if it makes your gyro and dterm traces cleaner. If you need the traces cleaner, lower the lowpass_hz. (don't go below 60/30, it will become unflyable below that)
Don't lower max throttle. Just calibrate your ESCs from 1000 to 2000, set maxthrottle to 2000 and minthrottle as high as you need for all motors to start (test and calibrate without props)

Addicted5, 2016-12-05:
i'm afraid its 2212.
But even without propeller i'm getting weird [Blackbox log](https://mega.nz/#!CMBXwaLB!OGgZ7QMy53fVnVmBIhOnHjZgA5pu7nH3V1yAJSf2-P0) & motor behaviour.
This is [cli dump](http://pastebin.com/Z49TZwiu).
I kept my quad quad on my table & caliberste both gyro & acc.All at mid except throttle which is constant at 1100.But you can see motor.idk wt fc wanna do & accelerometer feeling.

andre92, 2016-12-05:
I viewed your blackbox log and I found your problem: the accelerometer is bugging. Cleanflight 1.14.0 and 1.14.1 contain a critical bug that make accelerometer unusable. I see you are flying in level mode, which is heavily affected by accelerometer (sorry for not noticing earlier). 

You need to revert back to Cleanflight 1.13.0 until Cleanflight 1.14.2 comes out (where the bug is fixed) and use the following CLI settings:
gyro_lpf = 188hz
gyro_soft_lpf = 80
dterm_cut_hz = 40
acc_cut_hz = 10

Addicted5, 2016-12-05:
Thanks for info.
Flashed 1.13.
Seems like it's a lot better.
I'm getting a takeoff finally.
Time to go field.
Here is the [log](https://mega.nz/#!7BJjGJzJ!-2nfxPOfKPBk8CKEo5AHEdwLqUMZYf6tfmDRG01j45Y) & [dump](http://pastebin.com/znT77h8Z) after flashing 1.13 btw.Seems like 1.14 is buggy.
Mark as solved.

andre92, 2016-12-05:
From this latest log, it looks like you have a particularly noisy copter. Did you balance the propellers and soft mount the FC?
D-term is still amplifying vibrations. (and speeds motors up from idle to 80%) You can do 2 things to fix it: use lower D gains or apply more filtering. With the current filter settings, I would set D gain between 7-15, the default 23 is too high. Alternatively, you can lower dterm_cut_hz to 30 so it cuts the vibrations better.

Addicted5, 2016-12-05:
by soft mount, do you mean [this](http://img.banggood.com/thumb/large/upload/2015/03/SKU214860_4.jpg).
Propeller is balanced.
! thing i found.The seller send me 3 same size propeller adaper , 1 a bit smaller adapter.
Do i need to replace that smaller adapter?(smaller adaper making propeller mount bit higher i.e  0.3cm up than other)
& i noticed that motor takes off last when i increase throttle.

andre92, 2016-12-05:
yes, I used one of those for my 450 quad as well. If your frame doesn't have 3x3cm screw holes, you can always try to secure it with double sided tape and zip ties (just make sure it sits absolutely straight and center to avoid sensor offset)
Using washers should be ok although 0.3cm higher should not make it fly worse. 
As a side note, I found my 450 quad flies better with the battery strapped onto the top plate and the FC inside (if there is room). That also protects the FC if it falls upside down and it may help with the vibrations (some places on the copter are more sensitive to vibration than other)

Addicted5, 2016-12-05:
thanks to you i'm able to fly my quad.
I made the changes you suggested.Now battery on top, fc secured between 2 pcb of frame.
i would appreciate if you point some good tutorial to tune PID from scratch or tunning PID from blackbox log.

andre92, 2016-12-07:
try this one: https://www.youtube.com/playlist?list=PLwoDb7WF6c8ldO8tz0IUi9FNcJdvE2Mhe

hydra, 2016-12-11:
please test the 1.14.2 release candidate and report back.