cleanflight/cleanflight #2523: Why are 1.14.0 and 1.14.1 available as STABLE releases?

grizewald, 2016-12-03:
This is crazy!

New and inexperienced users don't go researching before they install what the configurator offers as the latest stable release, they just install it.

With the current horrible bugs that dramatically impact all self levelling modes and give excessive CPU usage on many targets, forums, reddit and facebook are filling up with people who can't understand why their craft won't fly properly.

Please, pull these releases or at least mark them as unstable.

Marmalade1PD, 2016-12-03:
That's exactly why I'm here.  
Previoius CC3D/Librepilot user trying CF for the 1st time, 'correct' orientations and I'm flipping on take-off despite only changing the release and using established settings.

grizewald, 2016-12-03:
The issues list isn't for support. RCGroups is a good place to find help.

Marmalade1PD, 2016-12-03:
I know.   I just saw a comment that I related  to

andre92, 2016-12-05:
another example. I can find nothing wrong here except his accelerometer #2522

Marmalade1PD, 2016-12-05:
I went back to the earlier version, no other changes.  Works great.

Thanks,

hydra, 2016-12-06:
@grizewald this is not the place for opening issues like this.  As developers we test and release code, sometimes bugs slip in, this is why we do hotfix releases.  We allow some time for people to test release candidates and report bugs, no-one reported the issue you were facing in the allotted time.  The issue you refer to was not an issue if your aircraft had good vibration isolation and hence it may not have been immediately obvious to the testers.

Please get involved with the release candidate process, try them and report back.

v1.14.2-RC1 has just been released.

closed 2016-12-06 by hydra