cleanflight/cleanflight #2524: PPM not worling on CC3D starting from version 1.13.0 (probably ppmCountShift is not set)

krzysztofkuczek, 2016-12-03:
Hi,
It looks like PPM is not working starting from version 1.13.0. I know that PPM input pit is changed from pin3 to pin8. 
I don't know cleanflight coding, but the problem should be related to not set ppmCountShift variable used in ppmEdgeCallback function.
As far as i see, it's default value is 0, an it could be set using ppmAvoidPWMTimerClash function.
But this function is called only once and only for SPARKY and ALIENFLIGHTF3 board.
So it could be like that there timer is not read correctly for cc3d boards...
Regards
Krzysiek

hydra, 2016-12-06:
any chance you could try updating the code, build binaries test and create a PR for us?

krzysztofkuczek, 2016-12-07:
yes, i could try to setup dev environment ;)
will let you know...

krzysztofkuczek, 2016-12-08:
Hi,
Just build binaries with code update, but it still does not work :(
Krzysiek

hydra, 2016-12-27:
unable to verify with v1.14.2 latest master from today.

steps to test:
connect PPM RX to Pin 8.
flash latest master full chip erase
enable PPM RX, save reboot.
all 8 channels working on receiver tab.
enable ONESHOT125, save reboot.
all 8 channels still working on receiver tab.

please test v1.14.2-RC3 when available and re-open if you still have issues.

closed 2016-12-27 by hydra