cleanflight/cleanflight #2525: Pitch and Roll drift SP F3EVO firmware 1.14.0 and 1.14.1

ecelkis, 2016-12-04:
using firmware 1.14.0 or 1.14.1 I am unable to use angle or horizon mode as pitch and roll drift over time.  Firmware 1.13.0 works well.

grizewald, 2016-12-04:
That's because both 1.14.0 and 1.14.1 are totally and utterly broken for any flight mode that uses the accelerometer.

The devs know about this and a fix has been committed, but since nobody gives a rats ass about flyers who use self levelling modes, nobody can be bothered creating a 1.14.2 release.

ecelkis, 2016-12-04:
Thanks grizewald,  I suspected that there was in fact an issue with the firmware...  I made the mistake of assuming that everything would work with a "stable" release.  Ugh!

andre92, 2016-12-05:
https://github.com/cleanflight/cleanflight/milestone/15
apparently, new hardware support is now more important than bug fixing

hydra, 2016-12-06:
@grizewald ahem!

I've been busy with the SPRacingF3NEO code and the UKDroneShow, fix already applied to the v1.14.2 branch. NEO code merged to master, release of v1.14.2 imminent.

@andre92 thanks for pointing the milestone out to @ecelkis and @grizewald 

Nothing more to do/say on this issue so I will close it now.

closed 2016-12-06 by hydra