cleanflight/cleanflight #2527: building in eclipse Window10: Program "" not found in PATH

pauapau, 2016-12-08:
hey, 
when i build with cygwin everthing is working fine. but not in eclipse
Mars.2 Release (4.5.2) CDT
GNU ARM ECLIPSE 2.8 (installed over https://github.com/gnuarmeclipse/windows-build-tools/releases)
GNU ARM ECLIPSE 3.1 (install over maketplace from eclipse)
GNU ARM EMBEDDED (installt over https://launchpad.net/gcc-arm-embedded/4.9/4.9-2015-q3-update)

i added  C:/cygwin/bin to PATH

what im doing wrong?

pauapau, 2016-12-08:
Solution:
i installed https://github.com/gnuarmeclipse/windows-build-tools/releases/tag/v2.8
then i copyed C:\Program Files (x86)\GNU ARM Eclipse\Build Tools\2.8-201611221915\bin to C:\Program Files (x86)\GNU Tools ARM Embedded\5.4 2016q3\bin

hydra, 2016-12-11:
thanks for posting your solution.

closed 2016-12-31 by hydra