cleanflight/cleanflight #2528: MSP_SET_MOTOR should have timeout

eziosoft, 2016-12-09:
I think MSP_SET_MOTOR has to have timeout. If the communication is somehow interrupted (app crashes etc) motors should switch off.
What do you think?

hydra, 2016-12-11:
what is your use-case?

eziosoft, 2016-12-15:
example:
user is testing motors with props attached and quad flies away disconnecting usb cable.
or user uses my app and bt connection is lost or something like that.
It will be nice that the motors stops when there is no communication, so having timeout is good option here.
It should work like MSP_SET_RAW_RC. When there is no message received for specific amount of time the motors should be turned off.

Or maybe it is already implemented like that and I'm missing something?

closed 2017-02-25 by hydra