cleanflight/cleanflight #2529: Crash: USART1 listed twice in CFC

emente, 2016-12-09:
Problem: 
- Cleanflight Configurator lists UART1 twice, see screenshot.

Symptoms:
- Changes on first UART1 (listed after VCP) are ignored (except data toggle).
- Changes on second UART1 are saved but seem to corrupt data. CF crashes after about 30 seconds, sometimes blinking slow and steady, mostly signaling 6 blinks, on very few occasions 2 blinks. 

Steps to reproduce: 
- freshly flash CF 1.14.1 on CC3D (F1 cpu) via main port
- disable data and enable serial_RX on second UART1, save, reboot
- observe irratic behaviour and crash

Versions:
Flight controller info, identifier: CLFL, version: 1.14.1
Running firmware released on: Nov 11 2016 11:55:21
Board: CC3D, version: 0
Configurator: 1.2.4

[image]

emente, 2016-12-11:
CF 1.12.0 (CFC 1.2.4) works correctly.

hydra, 2016-12-11:
have you tried flashing with full chip erase?  thanks for the report, not seen this issue before.  it does feel like the config is corrupt somehow, we just need to narrow down when this is happening.

emente, 2016-12-12:
Yes, flashed with full chip erase. It happens right after you change something on the (duplicate) UART1 and save. CLI also reports duplicate UART1.

As said above, it works with 1.12.0, no doubly detected ports.

GEORGEPLAVSIC180, 2016-12-23:
HOW TO ON WORKING THE CLEANFLIGHT ?

hydra, 2016-12-27:
ok, fixed. cause was this: e1ef91c0bc56b468fb2012d212a5f02524972ac2

fixed by a8ccc57445f192d104aff4e2e344228636135479
re-added support for SOFTSERIAL via 11546a1ba2261d987a6197d88f2627f98d8b3a1f

please test new release candidate RC3 when available.

closed 2016-12-27 by hydra

emente, 2016-12-30:
1.14.3 works correctly. thanks.