cleanflight/cleanflight #2532: CF 1.14.2 RC1 is not saving accelerometer calibration (Flip32 10DOF) after disconnect from USB power

de-o, 2016-12-13:
Configuration:  a loose Flip32 10DOF (uninstalled, on a bench), CF Config v. 1.2.4 via macOS 10.12.1, CF 1.14.2 RC1

With 1.14.2 RC1 (1.14) installed via CF Config 1.2.4, if I flash 1.14 to the flip32 then calibrate the sensors, then go to the sensors tab, I can see everything registering fine in the sensors tab of CF Config.  But if I disconnect the board and pull power (by disconnecting the USB), then reconnect via CF Config and go to the sensors tab again, the accelerometer is not registering.  If I recalibrate the sensors, then go back to the sensors tab, the accelerometer will register again until power is again disconnected.  

Flashing back to CF 1.13 on this same config does not reproduce this issue.

de-o, 2016-12-13:
I did a cli dump after calibrating 1.14.2 RC1 and saved the output to a file, disconnected from CF Config and then diconnected power.  I then reconnected and checked the Sensors tab and the Accelerometer graph was not registering changes.  Without recalibrating sensors, I did another cli dump to a file, then diff'ed it with the previous dump.

$ diff 1_14_2_RC1_pre.txt 1_14_2_RC1_post.txt
419c419
< #
---
> #

de-o, 2016-12-14:
A user on RCGroups tested this scenario with his SPF3 and was able to reproduce this also.

grizewald, 2016-12-15:
If I flash 1.14.2-RC1 to a full Naze32 Rev 5 (with full chip erase) and then calibrate the accelerometer, switching to the sensors tab shows the accelerometer to be working.
If I then enter the CLI and just type 'exit' to reboot the board, the accelerometer stops working.
Neither cycling power or resetting the board gets the accelerometer working again. Only a recalibration gets it going and that only lasts until the next reboot or power cycle.

CerealKiller1, 2016-12-18:
I am the RCG member de-o mentions testing on my SPF3,  I followed his steps and can reliably reproduce every time!

hydra, 2016-12-20:
thanks for the excellent and clear bug report.

hydra, 2016-12-21:
please try the latest master or v1.14.2 branch

grizewald, 2016-12-21:
I just flashed 4591_8ebc3bc_travis_cleanflight_NAZE.hex from cleanflight.memoryleaks.org onto my Naze32 and can confirm that the accelerometer continues to work after a reboot.

Selecting any other PID controller than LuxFloat still makes CPU usage go to 800% though.

CerealKiller1, 2016-12-21:
I tested 4591_8ebc3bc_travis_cleanflight_RMDO (obviously on my DODO) and as @grizwald found in his the ACC looks good after reboot!

I'm still using cofig1.2.3 so I cant see CUP usage but on all 3 PIDC's port utilization %'s remains the same and loop time is ~500uS and stable (as stable as possible)

de-o, 2016-12-22:
-Tested in master (275dd8a) and can confirm the acc fix
-e30e83e (Prevent serial overflow in BLHeli Passthrough) was also merged into master and I don't have an easy way to test this, other than a code review.  FWIW, the simple while loop looks sound to me.  
-version.h is still showing 1.14.1 so might want to bump that up.

Thanks for jumping on this fix, everyone! VERY much appreciated!

de-o, 2016-12-28:
Closing this issue, since a fix has been merged.

closed 2016-12-28 by de-o