cleanflight/cleanflight #2542: 'accSamplingInterval'  is used uninitialized, cause 'accSmooth[]' always zero!

DongChina, 2016-12-19:
accSamplingInterval <- biquadFilterInitLPF() <- accelerationFilterInit() <- activateConfig <- readEEPROM() <- init() .

andre92, 2016-12-19:
confirmed, level mode is utterly broken. 1.14.2 flies acro only.

martinbudden, 2016-12-19:
The underlying problem is that `accSamplingInterval` is defined in the header file, rather than declared `extern`. See

https://github.com/cleanflight/cleanflight/blob/master/src/main/sensors/acceleration.h#L40

andre92, 2016-12-19:
@martinbudden The only thing I noticed was accSamplingInterval being used uninitialized and I fixed that the same way Betaflight does. I only tested a single FC on the bench so far, but my PR did fix the issue of no acc traces on the sensor tab.
accSamplingInterval was set to 1000 in the configureScheduler(), but acc filter was initialized before that, with sampling interval of 0, breaking the filter and outputting 0.
Betaflight fixed that by not initializing the acc filter until accSamplingInterval is set, and then initializing the filter in the main loop.

martinbudden, 2016-12-19:
Yes, it's all a bit of a mess at the moment. `configureScheduler()` is definitely the wrong place to set`accSamplingInterval`.

I think the correct fix is to declare `uint32_t accSamplingInterval = 1000;` in `acceleration.c` and change the reference in `acceleration.h` to be `extern` (and of course remove the code that sets it in `configureScheduler()`.

I am not in favour of using lazy initialisation, ie an `accFilterInitialised` flag, since this just masks the underlying problem (as well as being inefficient).

hydra, 2016-12-21:
done in master and v1.14.2 now.  lazy initialisation is gone.

closed 2016-12-21 by hydra