cleanflight/cleanflight #2545: combine flysky ibus telemetry and serial rx on same port

unitware, 2016-12-20:
Flysky ibus telemetry is discussed here: https://github.com/cleanflight/cleanflight/issues/1125

Meanwhile, I'm working on combining serial rx and ibus telemetry on the same pin without any specific mods to the flysky equipment. I'll start this issue for that discussion to stay focused.

stronnag, 2016-12-20:
Does #2345 help?

unitware, 2016-12-20:
I'm combining the ibus ports of the FS-iA6B receiver with a diode and a resistor.

```
+---------+
| FS-iA6B |
|         |
| Ser RX  |---|<---\       +------------+
|         |        |       | FC         |
| Sensor  |--#==#--*-------| SerialTX   |
+---------+                +------------+
```

[image]

unitware, 2016-12-20:
Prereviews are welcome:
https://github.com/unitware/cleanflight/tree/combine-ibus-serial-rx-and-telemetry

I have borrowed some snippets from https://github.com/cleanflight/cleanflight/pull/2345 to be able to combine the serial port functions.

Enable with cli:
```
serial 1 1088 1115200 57600 115200 115200
feature RX_SERIAL
set serialrx_provider = IBUS
save
```

inode-, 2016-12-28:
@unitware can you give us more information about what kind of resistor and diode you used?

Thanks

unitware, 2016-12-28:
10kohm and something like 1n4148 or similar. I just scrapped some stuff in my drawers. 

Skickat från min iPhone

unitware, 2016-12-29:
Lets move further discussions to the pull request: https://github.com/cleanflight/cleanflight/pull/2554

closed 2016-12-29 by unitware

MotoMotoRo, 2021-12-08:
Can it be used without a flight controller?