cleanflight/cleanflight #2547: Naze32 Full HMC5983 not working with 1.14

DiGiC81, 2016-12-25:
This is my first quad build with the naze32 full rev-6 and i ran into a problem.

When i upgrade the firmware to 1.14 the onboard magnetometer stopped working on my naze32 full.
All was working just fine with 1.13

I downgraded the firmware to 1.13 again and it started working.

What am i doing wrong?

andre92, 2016-12-25:
To answer your question, you flashed 1.14 to the board. See [this link](https://www.rcgroups.com/forums/showthread.php?2789903-PSA-Don-t-flash-Cleanflight-1-14-0-or-1-14-1-if-you-use-self-levelling-flight-modes%21) 1.14 also happens to be unflyable in angle mode, which I suppose you use if you are worried about Mag. 
1.14 was released early to support SPRacing's new boards, and a few bugs were introduced, I suggest you stay on 1.13 until they have been fixed.

DiGiC81, 2016-12-26:
Hi,

Thanks for the quick reaction.

I fly in rate mode almost alle the time.

But i like to have everyting ok when i lift off.

I will stay with 1.13 for now.

Outlook voor Android<https://aka.ms/ghei36> downloaden

closed 2016-12-27 by hydra