cleanflight/cleanflight #2548: SPracing F3, CF 1.14.2, drifting PID value

trekki1978, 2016-12-26:
Hi,

I'm new to this forum but I'm testing different flight controller firmwares for weeks now. Since I updated my SPracing F3to version 1.14.2 my quadcopter shows an abnormal behaviour. 
The copter is standing on the desk, ACC and MAG are calibrated and show nearly 0,0.
The copter is armed via Aux channel and throttel stick is pushed to 1156.
Now I would expect the motors to spin constantly but they don't. 2 Motors spin faster and faster and the others spin slower. The blackbox log shows changing values of the PID_I(yaw), PID_I(roll) and PID_I(pitch) but heading shows that the copter doesn't move.

Is anyone having an idea what happens? First I supposed a hardware issue but for me it looks like a software problem. This also happens on Betaflight.

I will attache the blackbox log and the profile dumps from CLI:

[blackbox_log_2016-12-26_214114.TXT](https://github.com/cleanflight/cleanflight/files/673315/blackbox_log_2016-12-26_214114.TXT)
[dump profile.txt](https://github.com/cleanflight/cleanflight/files/673317/dump.profile.txt)
[dump rateprofile.txt](https://github.com/cleanflight/cleanflight/files/673316/dump.rateprofile.txt)

Thanks for your help.
Marco (trekki1978)

andre92, 2016-12-27:
Does it behave normal on 1.13? 1.14 is known to have bugs with accelerometer

trekki1978, 2016-12-27:
Thanks for your quick help. You never stop learning, even about these obvious things.
I have trimmed the sticks again and put in a higher value for the deadband entries to filter the fluctuations from the RX. I will have to try it flying, tomorrow (at daylight) and send an update.

closed 2016-12-27 by hydra

trekki1978, 2016-12-29:
Hi,

I've not tried 1.13, yet. But 1.14.2 works good for me. I'm just playing
around with the copter and try to learn as much as possible (before I will
build a bigger octocopter ;-).
Maybe 1.13 would work better but I hope that 1.14 will be as good as 1.13
with this feature. I also do not understand why code is changed and gets
worse than before?!

Thanks
Marco (trekki1978)